team401/2025-Robot-Code #1: Pheonix-6-Drivetrain

planet-exe, 2024-11-22:


PChild, 2024-11-26:
Can we try [TorqueCurrentFOC](https://v6.docs.ctr-electronics.com/en/latest/docs/api-reference/device-specific/talonfx/talonfx-control-intro.html#torquecurrentfoc) instead of voltage control for both motors on each module? Eg change [here](https://github.com/team401/2025-Robot-Code/blob/daa33039d3a2ba9a37046356f81fde9ad19f2a75/src/main/java/frc/robot/generated/TunerConstants.java#L40) and we'll need to tune gains of course.

sleepyghost-zzz, 2024-12-05:
Hi @jkleiber, this meeting @PChild showed Aiden and I the new advantage kit stuff for 2025. They have talon fx swerve drivetrain and it has what we are looking for with other stuff too. I'm going to try to use it and see how it goes. Heres a link to look at it: https://docs.advantagekit.org/getting-started/template-projects/talonfx-swerve-template/

jkleiber, 2024-12-05:
@sleepyghost-zzz seems good to me, keep us updated on how it goes

sleepyghost-zzz, 2024-12-05:
okie :)

sleepyghost-zzz, 2024-12-20:
@jkleiber just a quick update on how the advantage kit stuff is going. I think it is really good and has lots of features we can use. Currently I am working on implementing a field centric or robot centric toggle,  setGoalSpeeds method, and the use of DriveWithJoysticks.

closed 2025-01-20 by linglejack06