team401/2025-Robot-Code #4: Add elevator

aidnem, 2024-11-26:
**Purpose**
The purpose of this addition is to add an elevator subsystem and a mechanism and IO for the elevator.

**Project Scope**
- [ ] Add an elevator subsystem (we might rename this later when we add in the arm, since the arm and elevator will probably be mechanisms in the same subsystem).
- [ ] Implement an elevator mechanism that lets elevator be commanded to a position, and can return the elevator position to anything that needs it.
- [ ] Add a sim elevator IO that simulates a physical elevator and the CRT encoder values.
- [ ] Make elevator mechanism convert CRT encoder values.
Later:
- [ ] Add hardware IO that behaves the same as sim

PChild, 2024-11-26:
Can we try Motion Magic Expo for the elevator please? https://v6.docs.ctr-electronics.com/en/stable/docs/api-reference/device-specific/talonfx/motion-magic.html#motion-magic-expo

aidnem, 2024-11-26:
Sure!

PChild, 2024-12-08:
@jkleiber is someone assigned to do sim for this?

jkleiber, 2024-12-08:
@PChild I believe @aidnem is working on the sim for it

aidnem, 2024-12-08:
Yes, I'm working on sim. There's a basic version implemented right now but I think the gear math is wrong because some of the numbers don't make sense.

closed 2025-01-13 by aidnem