team401/2025-Robot-Code #11: Add elevator CRT failsafe

aidnem, 2024-12-21:
**Purpose**
If the elevator fails to seed with CRT (most likely due to backlash) we need the elevator to still be usable.

**Project Scope**
- [ ] Add LED indicators to warn when CRT isn't seeded pre-match
- [ ] Use this issue to discuss a potential solution. Current candidates are:
  - Use roboRIO user button or a DIO port to manually seed the elevator at 0, so that if CRT fails prematch we can manually move elevator to bottom and seed it that way
  - Pulse voltage on the elevator to remove backlash and allow CRT to find a solution
- [ ] Update this issue's description and implement the solution(s) chosen.

jkleiber, 2025-02-12:
@aidnem I'd recommend trying the median filter first if you go the filtering route. Kalman filter is probably not going to be very useful here. If the median filter doesn't work maybe we can try out a digital low pass filter https://docs.wpilib.org/en/stable/docs/software/advanced-controls/filters/linear-filter.html

jkleiber, 2025-02-12:
to clarify: a kalman filter would be very useful for estimating the state of the elevator (based on our encoder position and velocity measurements), but we would need to build a model for the elevator first and also think a little bit harder about stuff like our noise model of the encoder. We definitely can go this route, but I think we should start with a simple filter to see if that gets us enough performance for now

aidnem, 2025-02-13:
I'm prioritizing writing this over the tunable mode since they'll need to steal elevator to put wrist and claw on it, during which time I can throw together the test mode.

jkleiber, 2025-02-13:
Makes sense 👍

aidnem, 2025-02-13:
@jkleiber mind if I put this on the 74-tunable-elevator branch since it's sort of become our integration branch?

jkleiber, 2025-02-13:
I think that is acceptable, let's do it so we can test on hardware asap

aidnem, 2025-02-13:
@jkleiber I'm not sure what I/we did but it can't seed at all in sim now so I think we really broke something

aidnem, 2025-02-13:
Okay turns out commenting out all of your CANcoder configurations and half of your CRT logic and then also not handling encoder inverts properly *will cause issues*.

I've added some insane standard deviations to sim (0.1 meter std dev) and it can still seed after about a second, so I want to try on hardware again after I've fixed all of the obvious reasons that CRT couldn't seed.

aidnem, 2025-02-16:
@jkleiber I'm going to make there be a smaller threshold for the amount of time the elevator can NOT move before it considers itself at zero. This way, if the elevator is at the bottom and it starts homing, it won't have to wait the full 3 seconds or whatever we decide on as the maximum time before it realizes that it's at the bottom.

jkleiber, 2025-02-16:
@aidnem that makes sense to me

aidnem, 2025-02-16:
Here's verification of it working in sim:

[image]

It starts pushing the elevator down, waits until it stops, then seeds to zero and transitions to "Idle".

jkleiber, 2025-02-18:
I view this as complete via #83

closed 2025-02-18 by jkleiber

aidnem, 2025-02-18:
Agreed