team401/2025-Robot-Code #12: Determine if CRT can be wrong

aidnem, 2024-12-21:
**Purpose**
Figure out if backlash or measurement error can cause CRT to be incorrect and cause some error in the seed.

**Project Scope**
- [ ] Test seeding the elevator with CRT a ton of times, then dropping it to the bottom of its range of motion and seeing if there is any error. These tests should start at various heights, should happen while moving up, down, and not moving, under power, naturally falling, etc.
- [ ] If error is found, re-seed periodically or use another method (such as in #11).

jkleiber, 2024-12-21:
You could probably make a test mode in sim to set the position of the elevator (ground truth) and then simulate exactly what you're describing here over and over

aidnem, 2025-02-21:
We determined this: It very much can be wrong, and frequently was. As a result, we moved to homing instead.

closed 2025-02-21 by aidnem