team401/2025-Robot-Code #13: Add support for MapleSim in drivetrain code

jkleiber, 2024-12-24:
**Purpose**
The purpose of this addition is to enhance the overall robot simulation by incorporating [MapleSim](https://github.com/Shenzhen-Robotics-Alliance/maple-sim?tab=readme-ov-file) into the project.

MapleSim has physics based simulation natively and could be used to simulate game piece intake and scoring (things that we haven't done in the past). The goal of this ticket is to get the ball rolling and see if this is something we want to pursue further by integrating it with the drivetrain being added in #1 

**Project Scope**
- [ ] Add MapleSim to the project for the drivetrain
- [ ] Evaluate pros/cons of MapleSim
- [ ] determine path forward and development roadmap

**Verification**
- screen recording or demonstration of robot driving in MapleSim

linglejack06, 2025-01-05:
@jkleiber up to you, but I can take this on when I work on integrating drivetrain with Vision and drive with joysticks command. As for vision, I figured I'd integrate it quickly in code then spend time with Josh + maybe Aiden at first to get cameras calibrated and start testing on hitl.

jkleiber, 2025-01-05:
@linglejack06 I was thinking of this being a good task for @minhnguyenbhs actually, but we can see how it goes!

linglejack06, 2025-01-05:
@jkleiber thats great too I don't mind either way

minhnguyenbhs, 2025-01-05:
if you don't mind i'd love to at least help with this it looks really cool :D
is the drivetrain done atm?
should I add the high-key advantagekit cad model as well? I think @avidraccoon was able to fix the arm (I haven't gotten to testing it yet, sorry) and the drive-train won't need any additional moving components anyway

aidnem, 2025-01-06:
Drivetrain is close to being done as far as I know, Josh is working on it as we speak.

minhnguyenbhs, 2025-01-07:
oh nono sorry i wasn't trying to rush you, i was just a bit overexcited to get started haha! I definitely have enough stuff to work off of currently, and I'll rebase my branch off of yours when you're done

closed 2025-02-10 by avidraccoon