team401/2025-Robot-Code #15: Implement LED subsystem

jkleiber, 2024-12-25:
**Purpose**
The purpose of this addition is to implement the LED subsystem.

We should leverage new WPILib features to do this, read more in this PR: https://github.com/wpilibsuite/allwpilib/pull/6344

**Project Scope**
- [ ] Come up with LED light codes
- [ ] Implement LED subsystem
- [ ] Simulate LEDs
- [ ] Test on HITL

minhnguyenbhs, 2025-01-10:
Are there LED simulation capabilities on Glass/Advantagekit?

jkleiber, 2025-01-10:
@minhnguyenbhs there are ways to simulate LEDs in Glass yeah

minhnguyenbhs, 2025-01-15:
LED signals required:
- [ ] (rainbow) turned off/on
- [ ] climb locked on bars (bar LEDs)
- [ ] target found on reef (middle/bottom); specific additional cues for level chosen
- [ ] target found on coral source (middle/bottom)
- [ ] target found on algae processor (middle/bottom)
- [ ] target found on net (middle/bottom)
- [ ] holding algae (top left?)
- [ ] holding coral (top right?)

minhnguyenbhs, 2025-02-16:
@sleepyghost-zzz hi!! sorry i wasn't at shop yesterday :( do you want to keep working on this branch or are you busy with something else?

sleepyghost-zzz, 2025-02-20:
@jkleiber @minhnguyenbhs could someone go into more detail on the "target on net" "target on coral source" "target on algae processor". I am a little confused on what to put here. Thanks

sleepyghost-zzz, 2025-02-20:
When you say target is that the OTF target?

sleepyghost-zzz, 2025-02-20:
@jkleiber

closed 2025-03-07 by jkleiber