team401/2025-Robot-Code #16: Add switch for LEDs and Brake mode

jkleiber, 2024-12-28:
**Purpose**
Every year the robot has physical switches to turn the LEDs on/off and to turn brake mode on/off. Both of these switches should only be effective while the robot is disabled.

See 2024-Robot-Code's RobotContainer for an example. This is a Digital IO when used on the real robot, but we could make it a dashboard switch for simulation testing.

**Project Scope**
- [ ] Implement switch for LED on/off
- [ ] Implement switch for brake mode on/off
- [ ] Simulate to prove it works

minhnguyenbhs, 2025-01-10:
Do you know if there is a way to simulate leds in advantagekit or elastic?

sleepyghost-zzz, 2025-02-20:
What is brake mode?

jkleiber, 2025-02-20:
> What is brake mode?

@sleepyghost-zzz motors can have two neutral modes: brake or coast

If the neutral mode is coast, the motor spins freely if not under active control by the software. So if the robot is disabled and a motor is in coast, we can move whatever is attached to that motor by hand

If the neutral mode is brake, the motor does not spin freely even if it isn't being controlled. It is still possible to move during disabled, but substantially harder.

We want a switch that turns off brake mode on some of the motors so we can reset the robot by hand after a match

closed 2025-03-07 by linglejack06