team401/2025-Robot-Code #17: Ensure vision works properly with drivetrain developed in #1

linglejack06, 2024-12-31:
**Purpose**
Make sure vision does not cause robot to estimate itself as much faster than truth

**Project Scope**
- [ ] add vision into robot container
- [ ] test in sim without drive with joysticks command (rule out command)
- [ ] test in sim with command to see if everything works together
- [ ] fix whatever arises as the issue, or no fix is issue isn't found in this code

linglejack06, 2025-01-02:
@jkleiber @aidnem vision has been ruled out as issue with taon fx swerve template. My guess now is that it is something in regards to our drive with joysticks command interfacing with the talon fx swerve. I will spend some time looking over what they have done different. So far the only thing i notice is that they are calling ChassisSpeeds.fromFieldRelativeSpeeds before passing a speed to run velocity, which we arent doing.

This should be an easier fix though than if it was vision so that is good news.

linglejack06, 2025-01-02:
ill likely wait to mess with this more until i can actually touch the drive as written for 2025 though. No use in adjusting their template just to do it again to drivetrain for 2025. As soon as 2025 drivetrain is done though ill get this fixed

linglejack06, 2025-01-31:
this should be closed when single tag lineup is tested, already confirmed in sim

jkleiber, 2025-02-02:
It worked yesterday, both at localizing and at single tag lineup. Closing this ticket since follow on work will proceed in the relevant feature tickets

PR #56 (2025-02-04): Add final landing zone lineup at reefs
- add several methods relating to lineup: find tag id, find camera id, find goal speeds, etc
- add pid controllers for along track, cross track, and rotation in lineup
- add vision in initSubsystems
- add vision constants

Resolves #31 
Resolves #17
closed 2025-02-02 by jkleiber