team401/2025-Robot-Code #21: Calibrate Cameras

linglejack06, 2025-01-08:
**Purpose**
Calibrating our cameras with photon vision on hitl will allow us to be prepared for when we get the real robot in our hands

**Project Scope**
- [ ] calibrate cameras 
- [ ] add camera parameters in 2025-Robot-Code and instantiate vision from coppercore into RobotContainer

minhnguyenbhs, 2025-01-12:
@linglejack06 hi! i'm currently at shop, do you want me to do any hardware things while i'm here?

linglejack06, 2025-01-12:
You could get rio on hitl flashed to 2025, if you have the timeSent from my iPhone

linglejack06, 2025-01-31:
update: front right camera calibrated for 1280x800 and 800x600 on test_drivetrain

jkleiber, 2025-02-18:
I believe this is done enough to close now that we have all the HD cameras installed.

closed 2025-02-18 by jkleiber