team401/2025-Robot-Code #30: Add robot rotation lock on reef

linglejack06, 2025-01-14:
**Purpose**
The purpose of this addition is to be able to rotate around the reef easily

**Project Scope**
- [ ] Add pose of center of reef
- [ ] override rotation of goal speeds from joysticks when wanting to align to the reef

jkleiber, 2025-01-15:
@aidnem this ticket is for keeping our robot heading pointed at reef center. We'd use this during OTF and override heading target from OTF. Then we'd fully line up with #31

linglejack06, 2025-01-15:
@jkleiber regarding overriding heading. couldnt we just make final pose for otf against each side with the sepcified heading, then we woudlnt even have to hand over alignment to robot, especially since headings are pretty simple, (0,60,120,180,240,300) right? Aiden and I were talking about this yesterday. However, do you see any benefits to letting the robot control it instead of otf? 

my thoughts were otf could take our robot exactly in front of the reef poles, rather than just to a landing strip, unless we find otf to be inaccurate.

jkleiber, 2025-01-15:
@linglejack06 we'll want this mode anyway for teleop drive (we aren't going to be able to otf through defense)

It is also nice to be able to know what the robot heading will be during the otf. By controlling to reef centric heading the motion will likely be smoother. We can test to see what otf does by default though

linglejack06, 2025-01-15:
@jkleiber ok, that makes sense.

closed 2025-02-06 by jkleiber