team401/2025-Robot-Code #31: Add robot rotation lock to sides of reef

linglejack06, 2025-01-14:
**Purpose**
The purpose of this addition is to help auto alignment with scoring in teleop and auto progressed by otf path follow

**Project Scope**
- [ ] Add tag numbers with some sort of control to select them
- [ ] override rotation from joystick to lock onto specified tag id

aidnem, 2025-01-15:
@jkleiber @linglejack06 Do we want to lock onto the direction of the side of the reef instead of pointing directly at the tag? This way we'd stay aligned if we have to be off center to score (I'm not sure how the robot will look but I don't think we can stay 100% centered on the reef side and still be able to score).

Maybe we could do this only when we're within a certain distance to make sure that we stay pointed at the tag while far away.

Benefits are as follows
- Easier to align (just switch to a heading)
- On-the-fly can just be provided with an end goal heading to line up to the side of the reef
- Less rotation while strafing around the reef, resulting in less rotation while driving and therefore less drive skew (not sure if this was an issue last year or not)

jkleiber, 2025-01-15:
@aidnem yeah the idea of this ticket is to basically have a mode where we are lined up directly facing the relevant side of the reef for scoring purposes after we OTF over to a scoring position

sleepyghost-zzz, 2025-01-29:
@jkleiber hi, i was curious about 2 things. one, can i get an updated definition of what i should be doing, and 2, can you look at the (WIP) commit and help me figure out why it isnt turning?

linglejack06, 2025-01-29:
@sleepyghost-zzz fyi there's code for aligning to a specific rotation point in high key 2024 ( somewhere in drive) and in 2024 robot code (again in drive, I think periodic or run velocity equivalent )

linglejack06, 2025-01-29:
That code might help some, also check chief delphi for lots of examples

jkleiber, 2025-01-29:
@sleepyghost-zzz can you point me to a PR with your changes. It would make it easier to review

PR #56 (2025-02-04): Add final landing zone lineup at reefs
- add several methods relating to lineup: find tag id, find camera id, find goal speeds, etc
- add pid controllers for along track, cross track, and rotation in lineup
- add vision in initSubsystems
- add vision constants

Resolves #31 
Resolves #17
closed 2025-02-04 by linglejack06