team401/2025-Robot-Code #32: Add On the fly path following

linglejack06, 2025-01-14:
**Purpose**
The purpose of this addition is to have better autos and quick teleop cycles

**Project Scope**
- [ ] Add on the fly path following to coral station and each location on reef
- [ ] verify by running on the fly in auto and teleop
- [ ] verify ui controls with minh before merging

linglejack06, 2025-01-14:
follow docs here: https://pathplanner.dev/pplib-pathfinding.html#pathfind-then-follow-path

and here: https://pathplanner.dev/pplib-create-a-path-on-the-fly.html

and here for rotation override: https://pathplanner.dev/pplib-override-feedback.html

linglejack06, 2025-01-15:
@jkleiber @aidnem  regarding this issue, how would you like to set up commanding on the fly.

options i see are:
- have commands for each on the fly pose, these commands generate path then send speeds to drivetrainthrough setGoalSpeeds
- have command send on the fly pose to drivetrain, drivetrain creates path and follows it (this would likely require a state machine, which i assume we are using for alignment as well)
- have command poke state machine for specific side alignment, state machine in drivetrain then finds pose itself and gathers path that way.

Are we set on using a state machine for drive this year? i feel it might be easier just using commands, however, i know previously we have agreed that command scheduler makes this harder?

Ill work on this tonight at shop once we figure out specific option

jkleiber, 2025-01-15:
We'll need 24 OTF poses, ideally not hard  coded (either json or path planner defined)

I think state machine is the way to go to manage OTF vs. manual driving, then we'd just have specific heading modes that aren't state machine dependent (normal, reef centric, tag alignment)

For now, let's have the command pick a scoring index (0-11) and have the drivetrain figure that out based on alliance color.

jkleiber, 2025-01-15:
For the first attempt at this though we should prove the concept with just 1 pose

linglejack06, 2025-01-15:
> We'll need 24 OTF poses, ideally not hard coded (either json or path planner defined)
>
> I think state machine is the way to go to manage OTF vs. manual driving, then we'd just have specific heading [...]

@jkleiber okay that sounds good, would zero be the side facing barge or side facing alliance wall? and then ccw or cw from there?

so command structure would be: command puts drive in on the fly mode and tells drive a scoring location. then drive finds that pose and tells pathplanenr to generate path then follows it? as for this, drive would just switch back to manual once OTF finished? 

I also see a way where command could take over from DriveWithJoysticks then OTF command would check if drive is finished, once it is command would quit and DriveWithJoysticks could resume. This would remove need for OTF and Manual modes in drivetrain, but would require a isFInished supplier for command. Do yo just prefer the modes in drivetrain?

would heading mode for side alignment also be based on these scoring indexes? that way we would just have the one variable as i see no reason to align to a different side when we are trying to score on side 1 for example.

jkleiber, 2025-01-15:
@linglejack06 for side alignment we could have each scoring location affiliated with a side heading (like you described) as well as a tag offset for lining up

The one thing we'll need to think about more is that I think we'll want to be able to override OTF with the joysticks (i.e. we do OTF by default but the driver can take over to avoid defense). We'll need to figure out the interaction between drive with joysticks command and the OTF command

My first thought would be that the OTF command is an instant command that just sets the OTF target and pushes the state machine into that control mode (or whatever we decide to do there). Then, the drivetrain subsystem automatically sends itself to manual driving if it receives joystick inputs or if the path finishes it sends itself to scoring alignment mode?

We can do a lot of this in manual steps for now just to iron out the logic and see what works best for us

jkleiber, 2025-01-15:
Another thought I just had is to just extend DriveWithJoysticks and incorporate OTF there

linglejack06, 2025-01-15:
> @linglejack06 for side alignment we could have each scoring location affiliated with a side heading (like you described) as well as a tag offset for lining up
>
> The one thing we'll need to think [...]

@jkleiber i agree with figuring out things regarding the OTF Command, the issue i see is how we would determine when joysticks are giving input. Since all the drivetrain knows is what goal speeds are? 

I guess a solution could be to extend DriveWithJoysticks, but that might be too localized for coppercore.

A solution I could see is setting the interuption behaviour to cancel self on the OTF command, which would then set drive back to manual mode. I would just have to figure out how DriveWithJoysticks would be scheduled again and interrupt the OTF Command. Maybe this looks like checking inputs of joysticks in periodic and if they are true we schedule the joysticks command. What do you think of this workflow?

As for the scoring mode, once its in scoring mode, are we giving control of strafing to joysticks still, or will entire movement be controlled by a predetermined path (rotation and strafe)? I guess we could do something similar, where it follows path, with ability to override by using joysticks (rotation would still be controlled i imagine)

linglejack06, 2025-01-15:
example of OTF -> DriveWithJoysticks
```
 public void periodic {
    if(drive.isOTF()) {
       if (leftJoystick.getX() > 0.1 
             || leftJoystick.getY() > 0.1
             || rightJoystick.getX() > 0.1 // maybe we dont include this since rotation should pretty much always be handled
           ) {
                 Commands.run(DriveWithJoysticks, drive); // scheduling this would atuomatically cancel the OTF command
              }
     }
}
public void configureBindings() {
   // example of scheduling the command, this would require a non instant command to run the interrupt behaviour
   controller.X().onTrue(new OTF(1));

}

```

OR the other option i see is simply adding this code to DriveWithJoysticks

```
public void execute() {
  if (joystickInputsNotZero()) {
    drive.setMode(MANUAL)
    drive.setGoalSpeeds(...) // only set goal speeds here, we assume something else is controlling if we arent touching joysticks
  }
]
```

this way honestly seems simpler to me, the instant command would stop so drive with joysticks would take control again. with nothing happening in DriveWithJoysticks unless joystick inputs are non zero

jkleiber, 2025-01-15:
@linglejack06 I think the idea to extend DriveWithJoysticks makes the most sense. We'd keep the base coppercore functionality and then augment it with this game specific stuff. For example:

```
public class DriveWithJoysticksReefscape extends DriveWithJoysticks {

...
public void execute() {
  if (joystickInputsNotZero()) {
    drive.setMode(MANUAL)
    drive.setGoalSpeeds(...) // only set goal speeds here, we assume something else is controlling if we arent touching joysticks
  } else {
    super.execute();
  }
}
...

}
```

This is not exact, but maybe clarifies my idea some more

jkleiber, 2025-01-15:
To clarify: the `DriveWithJoysticksReefscape` would live in 2025-Robot-Code and would be how we also command auto-scoring most likely since it would require the drivetrain (via extending DriveWithJoysticks from coppercore)

> As for the scoring mode, once its in scoring mode, are we giving control of strafing to joysticks still, or will entire movement be controlled by a predetermined path (rotation and strafe)? I guess [...]

Long term goal is to have it be fully automated. But that shouldn't preclude the driver from doing manual stuff if they want to. For example, we could have a button that only scores but does not do any auto-align (though ideally this wouldn't be the default, but more of a fallback for if vision is failing or something)

linglejack06, 2025-01-15:
ok, i understand regardng the command, similar to my second idea just placed in command rather than in a periodic. 

that makes sense to me for scoring. first step for me now is probably just getting drive working with OTF commands. Then i enision next steps are aiming at center, aiming at side, then finally aligning strafe wise to score. I imagine many of these steps will involve @minhnguyenbhs for button decisions and @sleepyghost-zzz for some of the drive related tasks

jkleiber, 2025-01-15:
@linglejack06 that sounds like a good path forward to me! 

This ticket: get OTF drive working
#30: aim at center
#31: aim at reef sides
ticket tbd: align to score (we can add this later)

linglejack06, 2025-01-15:
ok, ill go ahead and get started on OTF tonight and should finish by saturday at the latest

jkleiber, 2025-02-02:
This work is definitely done, closing ticket

closed 2025-02-02 by jkleiber