team401/2025-Robot-Code #34: Figure out subsystems

linglejack06, 2025-01-14:
**Purpose**
Find out what subsystems we need to write and create issues for each of them

**Project Scope**
- [ ] talk to design about robot mechanisms
- [ ] turn those into individual subsystems after team talk

PChild, 2025-01-15:
# Robot details
Overall our robot will be very similar in function to the [WCP Competitive Concept](https://www.youtube.com/watch?v=n26NcMcYrFQ&pp=ygUXd2NwIGNvbXBldGl0aXZlIGNvbmNlcHQ%3D), please watch the video of it so that you have an understanding of how most of the systems will function.

## Subsystem specifics:

- <details open><summary><h2>Drive: moves the robot</h2></summary>
  
  - L1 (at least for now) Thrifty swerve
  - 6.75:1 reduction for drive
  - 25:1 reduction for steering
</details>

- <details open><summary><h2>Claw: Holds and scores both game pieces, rotates to different scoring angles</h2></summary>

  - <details open><summary><h3>Rollers: Index and score both coral and algae</h3></summary>
    
      - Powered by one Kraken
      - Probably 3:1 gearing reduction, not sure you all care though
      - We need some sort of sensor to know when we have a coral - do we want to buy a [CTRE CANrange](https://store.ctr-electronics.com/products/canrange?_pos=2&_sid=697875e04&_ss=r) for this?
      - We could (should?) auto turn off the ramp rollers when we sense that we have a coral in the claw
      - Do we need or want a sensor to decide when we have an algae? Could we do current or velocity sensing?
      </details>

  - <details open><summary><h3>Wrist: Rotates the claw to different positions</h3></summary>
      
      - Powered by one Kraken
      - Assume 20:1 gearing reduction for now, this may change
      - Default position will be the coral intake position at a hard stop
      - Will be able to rotate ~270 degrees
      - Will use absolute position of a CANcoder on the output shaft to determine wrist position
      - Several of the listed setpoints may be the same position but it's probably good to be able to control them independently
      - <details open> <summary><h4>Wrist setpoints needed:</h4></summary>

          - coral intake 
          - L1 score 
          - L2 score 
          - L3 score 
          - L4 score
          - reef algae intake 
          - ground algae intake
          - processor score
          - net score
          - climbing
        </details>
      </details>
</details>

- <details open><summary><h2>Ramp</h2></summary>
  
  Funnels coral to the claw.
  - <details open><summary><h3>Rollers: Pull coral down the ramp to the claw</h3></summary>

      - Powered by one Kraken
      - Probably 5:1 gearing reduction, not sure you all care though
      - Could maybe do current sensing for jam detection?
      - If it can jam it probably needs an unjam routine that pulses the rollers backwards and forwards
      </details>
  - <details open><summary><h3>Rotation: spins the ramp out of the way when we climb</h3></summary>

      - Powered by one Kraken
      - Probably 25:1 gearing reduction though this may change
      - For most of the match the ramp will rest against a hard stop in the down (coral intake) position
      - At the end of the match it needs to rotation the ramp ~270 degrees to get out of the way of the climb
      - We probably need to very lightly stall the ramp into the down position for feeding consistency
      - Preferably we do current or velocity homing against the down position hard stop and then just use the Kraken internal encoder for moving this to position at the end of the match, the pivot is probably dead axle so an encoder would be annoying and make design sad.
      </details>
</details>

- <details open><summary><h2>Climb: Lifts the robot off of the ground</h2></summary>
  
  - Probably two Krakens, may just be one, TBD
  - Gearing ratio probably 75:1 but not sure you all need to care
  - Functionally very similar to the [Everybot climb](https://www.youtube.com/watch?v=cM40ZVAj9Tk&pp=ygURZnJjIGV2ZXJ5Ym90IDIwMjU%3D) and [2582](https://www.youtube.com/watch?v=8MMuOQkpGa8&pp=ygUOMjU4MiBmcmMgY2xpbWI%3D).
  - There will be a passive latching system to grab the cage bars and then pivot the cage to cause the robot to climb.
  - Do we need or want inductance sensors to determine if we have grabbed the cage bars?
  - Needs to do position control to determine when we've rotated the cage enough to be in the air, this can be a CANcoder on the pivot axle.
</details>

- <details open><summary><h2>Elevator: moves the claw to different heights</h2></summary>
  
  - Two Krakens, probably on 5:1 reduction
  - CRTcoder for position sensing
  - Full down when intaking coral to save energy
  - Max height is still a bit TBD but we will give you one *soon*
  - <details open> <summary><h4>Needs a lot of setpoints just like the wrist:</h4></summary>
 
    - coral intake 
    - L1 score 
    - L2 score 
    - L3 score 
    - L4 score
    - reef algae intake 
    - algae ground intake
    - processor score
    - net score
    - climbing
  </details>
</details>
     
     
## Proposed CAN IDs, fuses, and ports for devices:
[image]
Click the picture to go to the live doc.

linglejack06, 2025-01-15:
[image]

linglejack06, 2025-01-15:
[image]

aidnem, 2025-01-20:
@jkleiber @PChild I feel like using a CANrange for detecting when we have an algae will be easier and more reliable than detecting it with current or velocity, but I've never tried using current/velocity to detect an object in rollers before. Do either of you have a preference?

jkleiber, 2025-01-20:
CANRange.

aidnem, 2025-01-23:
@linglejack06 I made some simple hitl test code on the [canrange-hitl branch](https://github.com/team401/2025-Robot-Code/tree/canrange-hitl).

Just ID the canrange with ID 30 and then enable test mode, it should output to SmartDashboard/canrangeTest/objectDetected and SmartDashboard/canrangeTest/distance

linglejack06, 2025-01-23:
@jkleiber do you have phoenix pro license account, I need to license the test bot motors

linglejack06, 2025-01-24:
@aidnem @jkleiber can range hitl works, its turning from false to true around a distance of 0.05 (i think in meters) so 5 centimeters. if my hand is anywhere above that object detected is false. does that seem right aiden?

aidnem, 2025-01-24:
@linglejack06 I configured it to detect at a range of 10cm but as long as its consistent, we can just tune it to work when we want it to.

linglejack06, 2025-01-24:
Ok, it might also just be distance outputted is a different unit, who knows

linglejack06, 2025-01-24:
@jkleiber @aidnem tomorrow, do we want to start getting vision calibrated so we can test otf and have it ready for single tag lineup

jkleiber, 2025-01-24:
@linglejack06 @aidnem yeah let's aim for that. Ideally we could have someone calibrating vision on HITL and another person getting drive working on the test chassis.

aidnem, 2025-01-25:
There will also be an intake arm that swings down. I could see this being part of scoring or its own subsystem, its up to interpretation. Lets decide on who to give this to during shop today.

PChild, 2025-01-26:
Checking the camera FOVs with some of the wrist stuff in place there will be portions of each view blocked when the elevator is all of the way down:

[image]

This is with the camera lens at exactly the height of the center of the tags and positioned such that horizontal tag center is on lens center when the robot is at either coral scoring position. 

Should we move the cameras to prevent occlusion? If so what's the preferred direction to move the cameras? Down? Further out to the sides?

jkleiber, 2025-01-26:
@PChild further out to the sides is my take. I'd prefer to keep vertical distance difference as close to 0 as possible

PChild, 2025-01-26:
The other option is we could maybe just not care? This only happens when the elevator is fully down, as soon as you raise it 1.25" the wrist stuff is no longer in view:

[image]

jkleiber, 2025-01-26:
@PChild if we only have to raise it 1.25" I think we're fine. We can try the design as is and move cameras if there is actually a problem

PChild, 2025-01-26:
Looks like the depth inset we did works for still seeing the tag when bumpers are against the reef wall:

[image]

aidnem, 2025-01-26:
We could just idle the elevator at 1.25" instead of 0, and then drop it to intake.

linglejack06, 2025-01-27:
@aidnem thats probabyl smart, also, when lining up to the reef, we should just go ahead and command elevator to desired level. which would resolve the camera issue there at least

aidnem, 2025-01-28:
@linglejack06  Yeah that makes sense we can enter prime when we enter lineup. Then the only time cameras are blocked is when we're actively intaking from coral station.

linglejack06, 2025-01-30:
@jkleiber do we want to use max resolution for cameras? photon vision says max can sometimes lead to performance issues on coprocessor, maybe we drop to like 800x600?

jkleiber, 2025-01-30:
We can try 800x600 at first, but higher resolution will let us see from further away

linglejack06, 2025-01-31:
@jkleiber 1280x800 calibrated with 0.31 mean error. 800x600 calibrated with 0.24 mean error. testing lineup now

jkleiber, 2025-02-13:
closing this since the work has been split out into separate tickets. Information can be found here if needed

closed 2025-02-13 by jkleiber