team401/2025-Robot-Code #36: Basic Functionality

minhnguyenbhs, 2025-01-15:
**Purpose**
Create basic functionality in Climb- io file, motors moving backwards, etc.

**Project Scope**
- variables:
- [ ] locked onto bars
- [ ] goalVoltage/reportedVoltage
- [ ] angle?
- [ ] current
- methods:
- [ ] lock onto cage
- [ ] pivot climb backwards
- [ ] home

minhnguyenbhs, 2025-01-18:
[image]
LOOK AT THIS FUN NEW ERROR

avidraccoon, 2025-01-19:
Is this for simulating the robot?

aidnem, 2025-01-19:
Sometimes sim fails to launch I think, sometimes I get that and just have to relaunch
I'll look up why it happens if it's actually making you unable to sim

minhnguyenbhs, 2025-01-19:
yeah this issue is for sim instead of getting the motors to work :(since no robot atm :D)
It resolves after being relaunched haha I just thought it was really odd- is it a new 2025 issue or has it existed since before?

aidnem, 2025-01-20:
I think I ran into it last year occasionally but I'm not sure. It could be because the laptops are so slow that it times out before it can launch or something.

avidraccoon, 2025-01-21:
It's a networking error with it being unable to make and connect to the needed port. I you forget to properly close the sim it sometimes happens.

closed 2025-02-13 by minhnguyenbhs