team401/2025-Robot-Code #38: Add claw mechanism

aidnem, 2025-01-16:
**Purpose**
Add a claw mechanism to the code so that we can control the claw.

**Project Scope**
- [ ] Add claw mechanism to the scoring subsystem
- [ ] Give it some sort of sensor to detect a coral
- [ ] Let it control one kraken for the wheels

aidnem, 2025-01-24:
@jkleiber do you have anything against giving subsystems a passthrough for their mechanisms, and mechanisms a passthrough for their IO?
I'm implementing the state machine right now and it seems that there's a lot of boilerplate for passing through values (e.g. the ClawIO implements isCoralDetected and then ClawMechanism has to implement isCoralDetected that just returns io.isCoralDetected() and then ScoringSubsystem has to implement isCoralDetected which just returns clawMechanism.isCoralDetected()).

With passthrough it would just be
```java
  scoringSubsystem.getClawMechanism().getIO().isCoralDetected()
```

Part of me doesn't want to do this because responsibilities should be separated from a moral standpoint but another part of me doesn't want a ton of boilerplate methods that don't do anything.

Do you have an opinion on the issue?

jkleiber, 2025-01-24:
@aidnem I think that with this level of nesting we need to be able to passthrough like you've described in order to remain efficient. I don't think this approach is unreasonable since each layer of abstraction still manages itself.

We just need to be careful that we aren't abusing passthrough to do stuff at the wrong layer of the code

aidnem, 2025-01-25:
I found a solution! I'm putting mechanisms in their own package (e.g. frc.robot.subsystems.scoring.elevator) and then making the control methods from IOs protected. This way the mechanism can control the IO but nothing outside of the mechanism can. All of the sensor readings will be public so that they can be read by anybody.

aidnem, 2025-01-25:
Nvm you can't protect interface methods. I'll figure this out later and trust myself not to do anything dumb in the wrong place until then

avidraccoon, 2025-01-28:
Could you not make It an abstract class?

closed 2025-02-02 by aidnem