team401/2025-Robot-Code #42: Convert elevator to scoring subsystem

aidnem, 2025-01-16:
**Purpose**
Convert elevator subsystem into the scoring subsystem by adding a wrist and claw.

This issue will contain multiple sub-issues.

**Project Scope**
- [ ] Add wrist mechanism #41
- [ ] Add code to limit wrist and elevator position's based on each other's positions: above a certain point, don't let let the wrist rotate out above a certain height, and don't let the elevator go above that certain height when the wrist is rotated out above the certain amount.
- [ ] Add claw mechanism #38

aidnem, 2025-01-28:
@jkleiber Potential scoring state machine:
Idle (holds at 1.25" up) -> can transition to either intake state or to a warmup state
IntakeCoral -> control to coral station setpoint, run claw rollers. transitions back to idle when a coral is detected
GroundIntakeAlgae -> move into position to ground intake an algae and run the claw and ground intake motors, transitions back to idle when an algae is detected
ReefIntakeAlgae -> move into position to intake an algae from the reef and runs claw rollers to intake. **TBD**: How do we decide which height to take it from (this is a user interface question). I'm thinking either we reuse the height/face selection or we make another one just for intaking. Transitions back to idle once claw motors are run.

ReefWarmup -> controls elevator and wrist to the reef setpoint (scoring will have a static field to hold the current level target) and waits. If force score is pressed, or once drive is in the right place (**TBD**: not sure where to put the logic to detect this yet) it can score.
ReefScore -> Same setpoint as ReefWarmup except it also runs the claw motor

ProcessorWarmup -> gets into position to process, and waits.
ProcessorScore -> same as warmup except it runs the claw to spit out. **TBD** How do we know when to transition to score when autonomous?

Barge warmup + barge score are identical to processor warmup / score.

Unknowns: [I'll write later, class just ended and have to catch the bus]

jkleiber, 2025-01-28:
@aidnem After last year's state machine stuff, I think we should try to have a really general state machine and use additional information from a command to decide which setpoints we are going for

Below I outline some states, and the metadata that can help the subsystem interpret what to do:

**States**
Idle
Exit criteria:
- Any valid command

Intake
Exit criteria:
- The desired GamePiece is in the robot
- Command canceled

Warmup
Exit criteria:
- We are actively commanding score
- All setpoints have been achieved
- Command canceled

Score
Exit criteria:
- The desired GamePiece is not in the robot
- Command canceled

**FieldTarget**
- Reef L1
- Reef L2
- Reef L3
- Reef L4
- Net
- Processor

**GamePiece**
- Coral
- Algae

**Actions**
- Intake
- Warmup
- Score

**Example flow**
Situation: full cycle for one Coral in teleop assuming a medium level of automation. Subsystem state will be shown as [State]

[Idle] we are driving around
[Intake] Driver presses intake button: IntakeCommand runs -> set action = Intake, set gamePiece = coral
[Intake] Robot positions scoring subsystem to intake coral at the coral station, other subsystems do required actions to intake
[Idle] Coral detected by CANRange -> subsystem detects game piece is acquired and goes to idle. intake action becomes invalid because we have a coral, so the state machine prevents transition 
[Idle] Driver begins driving to the reef
[Warmup] Driver asks for L4 position 2, masher sets position and driver commands warmup. WarmupCommand runs -> set action=warmup, gamePiece = coral, fieldTarget=ReefL4
[Warmup] Arm extends up to required height to score on L4, other subsystems do required actions (i.e. the drivetrain aligns or is manually aligned)
[Score] Robot is lined up, driver presses score. WarmupCommand canceled in favor of ScoreCommand runs -> action = score, gamePiece = coral, fieldTarget=ReefL4. State machine goes to the score state
[Score] Robot begins to eject game piece, CANRange can no longer see the coral
[Idle] CANRange has no coral for X timeout, arm descends back to idle position. score action no longer valid due to not having a game piece, so state machine stays in Idle

This way we can write commands that are able to orchestrate multiple subsystems, while keeping the I/O simple. We're basically creating an API of what the command can request from the subsystem here, and the subsystem should just take care of it.

The logic would be pretty much pulling setpoints based on the action + gamepiece + fieldtarget combos from JSON based lookup tables

aidnem, 2025-01-28:
Okay, this seems better than what I was thinking of lol. I am a big fan of having fewer states and more setpoints.

Thanks!

aidnem, 2025-01-29:
@jkleiber claw logic is pretty much done. Do we want to use masher trigger and bumper again for warmup and score until we decide on something else?

aidnem, 2025-02-03:
I realized I majorly misinterpreted the entire point of using an Inputs object for IO/replay mode stuff, so I'll have to do a minor refactor of elevator and claw so that all of the IO stuff is done through the proper objects so it can be replayed properly if we implement replay mode.

jkleiber, 2025-02-13:
Complete as of #41

closed 2025-02-13 by jkleiber