team401/2025-Robot-Code #48: Add support for multiple target platform parameter loading

jkleiber, 2025-01-22:
**Purpose**
The purpose of this issue is to create a system for determining which constants to use based on target platform.

Target platforms:
- comp bot
- programming bot
- HITL 
- Sim

I'm thinking one JSON file to pick between these options, and then a folder for each option with parameter JSONs

This way we can have comp bot with X feature flags/params, HITL with Y feature flags/params, and programming bot with Z feature flags/params and we never have to recompile or do anything weird to manage the parameters.

**Project Scope**
- [ ] Add a parameter JSON for picking target platform
- [ ] Structure parameter constants directory as described above
- [ ] Update code to pull in correct constants file path based on the target platform parameter JSON

jkleiber, 2025-01-22:
@avidraccoon could you take this on? It's very high priority so if you can't that's okay

avidraccoon, 2025-01-22:
I already have something mostly done for it

aidnem, 2025-01-22:
@jkleiber @avidraccoon Could we make a set of default configs that then only overrides the fields that are defined by each platform specific set of constants? E.g. the number of ticks on a CANcoder doesn't need to be robot-specific, but if for some reason it needs to change it would be better to only have to change it in 1 place.
This way each constants file would be small and it would be easier to keep them as similar to each other as we could.

jkleiber, 2025-01-22:
@aidnem imo stuff like that should be in a common JSON file that always loads regardless of platform. So yeah we should make that happen.

There will be some amount of duplication we have to accept though.

aidnem, 2025-01-23:
@jkleiber I've fixed all of the errors, but at the moment it seems to always read from the default and never from the correct path. I'm unsure why this is. After I fix that it will be ready.

aidnem, 2025-01-23:
it was another string comparison bug! However, now DriveWithJoysticks crashes because an argument to addRequirements is null for some reason. I'm unsure why. Once I fix *that* hopefully this will be ready.

aidnem, 2025-01-23:
It's not my fault, we can't feature flag drive out without a null pointer exception right now. I created a ticket and now this should be done.

closed 2025-01-23 by aidnem