team401/2025-Robot-Code #62: Add state machine to drive subsystem

jkleiber, 2025-01-30:
**Purpose**
The drive subsystem logic is getting complicated and would benefit from a state machine. After we get the whole scoring sequence to a MVP state, we should migrate over to using the coppercore state machine to make it harder to mess up the logic in the future (adding structure will make it easier to add new features).

In particular, the following states should exist:
- Manual
Edges:
- Command a reef target close enough to the target (-> Lineup)
- Command target position (-> OTF)

- OTF
Edges:
- Command canceled (-> Manual)
- Arrived at target + no lineup required (-> Manual)
- Arrived at target + lineup desired (-> Lineup)

- Lineup
Edges:
- Command canceled (-> Manual)
- New target commanded (-> OTF)
- **Note**: Lineup complete remains in lineup because we still are waiting to score

Driver Assist logic will live outside the state machine, and may be overridden in the OTF/Lineup states.
For example: always pointing at the reef logic may be overridden in Lineup state, but not in the OTF state. In the OTF state, this logic may be enabled when it would have been disabled in the Manual state by the driver. 

Driver Assist Logic:
- Point at reef center
- Align to net
- Align to processor

**Project Scope**
- [ ] Implement proposed state machine (+ any recommendations / findings from implementation)
- [ ] Create new files for state logic, and move state-specific stuff in Drive.java to the relevant state (See ScoringSubsystem for example)
- [ ] Ensure there are no regressions

linglejack06, 2025-01-30:
so basically we just add DriveState file between the drive and commands, ie commands poke drive state, drive state does its thing and tells drive what to do? this makes sense, but in my opinion this just adds complication to drive as we then have to worry about more complicated edge cases. Whereas now it just works by transitioning to lineup once OTF is close enough, rotation would be enabled during otf and cancelled for lineup. @aidnem what do you think? ill do whatever, once we get lining up etc to a desired functionality.

jkleiber, 2025-01-30:
This will accomplish a couple of things:

1. Make the Drive.java file less huge
2. Make the transition between various states more explicit

Point 1 is nice because then you don't have to read 700+ lines to find what you're looking for. If you want to see the OTF logic, got to the OTF file, and if you want to see the lineup logic you can go to the lineup file. This doesn't require a state machine to do, but the way our state machine is implemented encourages this already good behavior. So this is more of a "drive-by" fix.

Point 2 is nice because then we enumerate the transition logic and can't accidentally change it with commands/hacks at comp. This is my main concern. I'm willing to accept some additional complexity if that complexity guarantees reliable performance. In this case I don't think the overhead is that bad

jkleiber, 2025-01-31:
@linglejack06 yeah that's what I'm thinking

linglejack06, 2025-01-31:
@jkleiber ok, ill get started once we test and merge lineup to tag

jkleiber, 2025-02-10:
This was completed in #70

closed 2025-02-10 by jkleiber