team401/2025-Robot-Code #66: Add ground intake

jkleiber, 2025-02-03:
**Purpose**
The purpose of this addition is to add a ground intake subsystem 

This isn't super well defined yet so a significant portion of this issue will be figuring out how the subsystem works. 

I expect that this will be used to intake algae from the ground, but there will also be interactions when scoring in the processor

**Project Scope**
- [ ] Figure out what this looks like by talking to the design team
- [ ] Implement subsystem / mechanism logic for ground intake
- [ ] Add basic simulation of ground intake

minhnguyenbhs, 2025-02-17:
@aidnem hi!! would this go in the scoring subsystem do you think or will it just interact with it

jkleiber, 2025-04-14:
not happening anytime soon, closing

closed 2025-04-14 by jkleiber