team401/2025-Robot-Code #67: Tune lineup gains for fastness

linglejack06, 2025-02-04:
**Purpose**
The purpose of this addition is to make the final lineup with the tag happen faster

**Project Scope**
- [ ] tune gains in test mode
- [ ] add the encoder offsets, module positions, and other constant fields from test_drivebase  to comp drivetrain json too

jkleiber, 2025-02-10:
@linglejack06 before we commit to recomputing a motion profile every cycle to make this faster, I think we could try the following:
```
double maxVelocity = 3.0; // this is already set in constants somewhere
double maxAcceleration = 2.0; // this is already set in constants somewhere

// use vf^2 = vi^2 - 2ax to compute vi -> vi^2 = vf^2 + 2ax, vf^2 = 0 so vi^2 = 2ax
// x: alongTrackDistance
// a: maxAcceleration
// This assumes we are always decelerating. Since we have a max velocity we can clamp the result for vi
// to that value until we get below max velocity (when the equation tells us we need to decelerate)
double velocityCommandRaw = Math.sqrt(2*maxAcceleration*alongTrackDistance);
double clampedVelocity = Math.min(velocityCommandRaw, maxVelocity); // this is what we command
```
If we are at a slow speed and far away, we would send maxVelocity to the robot, but based on the current parameters that would not happen unless we are over 2.25 meters away. 

If we start at 0 speed we would always remain at or below the speed profile produced here and would have no issues, but it would be slow.

If we start above the recommended speed by this profile we could theoretically have issues (say we're going 3 m/s initially and then this says we should only be at 1 m/s because we are 0.25 meters away, we would crash into the reef pretty hard). To mitigate this, we should improve the OTF handoff logic some

The OTF handoff logic needs improvement anyway because waiting to get to the exact point for a reef target slows us down. We should check instead that the desired tag is in view and we are slower than (or close enough to if above) the speed target recommended above for lineup. That way we can transition with speed into lineup

linglejack06, 2025-02-10:
@jkleiber this could make sense, not sure how we would implement the otf handoff logic though (would i just let otf leave otf once vision has a distance error relative to its tag?

The issue i see with that is that all of that logic is centered in the lineup state, so this would require copying over those functions into otf.

Another option is we can just set a goal end velocity of 0.5/1 on otf path (this prevents robot from fully stopping then speeding up again for lineup) 

the poses where they are now are about the farthest they can be cross track from the tag in order to still see the cameras. I tested with more norm error allowed (otf pose exact vs robot pose ) this led to lineup failing as it didnt see a tag

jkleiber, 2025-02-10:
If the poses are already good and the limits are already good, then the simplest solution is to set a desired exit speed (preferably 1m/s to start, and 100% of that velocity would be facing the reef). We will need to make this a JSON constant

jkleiber, 2025-02-18:
Initial statement of work completed in #84 

Closing this ticket. I will open a ticket for additional optimizations

closed 2025-02-18 by jkleiber