team401/2025-Robot-Code #77: Make auto sequence work in sim

jkleiber, 2025-02-10:
**Summary**
Auto in simulation skips the lineup state after the first try, so I suspect there is some stale state hanging around that is short circuiting the lineup state logic.

For AprilTags: I think issues in #72 related to not being able to see the tags are due to using the right camera for the left camera's job. We should add support for the left camera in sim in this ticket.

we will be using sim a lot in the coming days/weeks to prepare for auto, so let's make sure the sim is as accurate as possible.

**Work Scope**

- [ ] Determine why the lineup state is being short circuited after the first score in auto
- [ ] Add the left camera to sim to prove it works (we can also add it to the robot here if convenient)
- [ ] Check that the apriltags are being used correctly in sim

linglejack06, 2025-02-10:
@jkleiber lineup is only skipped for the first reef on red side. after testing, its just because the camera isnt seeing the april tag in front of it (so it's pushing back to joystick state)

i assume adding left camera to sim might fix this, but unsure why it isnt seeing the tag in the first place. when switching to blue alliance, the auto runs perfectly (with lineup happening at both locations)

jkleiber, 2025-02-28:
@minhnguyenbhs after you get the maplesim game piece stuff in a decent state + merged, I think this would be helpful to work on. I haven't seen much traction on this since the team is mainly focusing on robot integration, so I think you're in the perfect spot to make this happen.

The issue as reported is that auto isn't working in sim because photonvision sim isn't showing the tag to the camera when the camera should be seeing it. This isn't technically critical path, but it would be really really nice to have

There are some obvious things to check first such as camera FOV settings being correct in sim. Also, if we are seeing too many tags, we could probably use maple sim's true robot position to find which tags are actually in view (we can't see tags through obstacles). It would be cool to see the 3D robot score 3D game pieces in sim as well

This would also involve making some "human player simulation" that feeds a coral to the robot whenever it goes to the coral station. Maybe you already have this in your other PR; if not, it can just be something hacky that only runs in sim to give the robot game pieces (ideally with some configurable delay because human players can't instantly give robots game pieces)

jkleiber, 2025-02-28:
@minhnguyenbhs I think it could be as simple as "if robot near station for > 0.5 seconds, spawn a coral into the robot's intake"