team401/2025-Robot-Code #82: Determine scoring setpoints

aidnem, 2025-02-12:
**Purpose**
The purpose of this addition is to determine all of the setpoints from ScoringSetpoints.java and the base min and max angles/positions of the wrist and elevator

**Project Scope**
Determine values of the following:

- [ ] Wrist min angle
- [ ] Wrist max angle
- [ ] Min elevator height
- [ ] Max elevator height
 - [ ] Setpoint L1
 - [ ] Setpoint L2
 - [ ] Setpoint L3
 - [ ] Setpoint L4
 - [ ] Setpoint L2algae
 - [ ] Setpoint L3algae
 - [ ] Setpoint coralStation
 - [ ] Setpoint processor
 - [ ] Setpoint net
 - [ ] Setpoint ground

jkleiber, 2025-02-20:
@aidnem cracked idea, but we should be able to do this fairly accurately in sim once 3d mode is merged. I'd like to see us try that so we have numbers that are at least close going into a tuning effort on the real robot

aidnem, 2025-02-20:
I'll try that out.

Only problem with that is the wrist angle (zero is in a weird and wacky place due to center of mass). I'll use the center of mass that 
CAD gave me and hope that it's pretty accurate to real life.

jkleiber, 2025-02-20:
This would be a good opportunity to make sure our transforms are right so this is consistent between sim and real life as well

jkleiber, 2025-02-22:
This ticket is done enough to move off the auto milestone, so I'm moving it to Competition 1 Readiness

jkleiber, 2025-02-23:
@avidraccoon I heard you got scoring setpoints for algae tonight - could you push those?

avidraccoon, 2025-02-23:
Don't have them in a constants file will add them tomorrow from the photos

aidnem, 2025-02-28:
We've redetermined these with the new claw now

jkleiber, 2025-03-04:
Moving to Blacksburg milestone because we can score all the game pieces everywhere for now

jkleiber, 2025-03-04:
For portsmouth L1 doesn't matter because it isn't strategically relevant

We can cook something up on the practice field if needed

closed 2025-04-14 by jkleiber