team401/2025-Robot-Code #88: Root cause and fix bug causing OTF to not always activate

jkleiber, 2025-02-13:
**Purpose**
The purpose of this ticket is to investigate and fix a bug involving the pose of the robot being co-located with the OTF target despite the robot not being close to the target at all

Suspicious code:
- getPose() in Drive.java
- check that the state machine isn't doing something weird

**Project Scope**
- [ ] Figure out the problem's root cause
- [ ] Implement the solution

aidnem, 2025-02-20:
@linglejack06 have you figured this out yet? Once we get scoring setpoints taken care of we'll need this to be fixed in order to run autos.

linglejack06, 2025-02-20:
No, but this isn’t an issue In auto, only a very niche case in teleop. It’s very hard to discern why @aidenm

aidnem, 2025-02-20:
Okay okay I see. If it's teleop only I feel like we just tell driver to pull the trigger twice right?

linglejack06, 2025-02-20:
Yeah that’s likely what we should do, I can try to fix it though as well just for convenience

PChild, 2025-02-20:
My 2c, we still have two weeks until competition and this may have a root cause that affects other systems (if it is a state machine bug for example...) so we should spend the time understanding and fixing it correctly. Telling the driver to pull the trigger twice is not a fix, that's just masking the problem and is really only acceptable if we get to day of comp and haven't fixed this.

aidnem, 2025-02-20:
Oh I agree, I'm not saying we shouldn't fix it, I'm saying that if we can't fix it it isn't the end of the world. I think that while part of controls works on getting scoring setpoints on comp bot, another part of the group should be trying to fix this on test drivebase or in sim.

jkleiber, 2025-02-28:
@aidnem / @linglejack06 do we think this issue is resolved now?

aidnem, 2025-02-28:
If I recall correctly, it still did the "dont move and only turn" behavior at shop yesterday but I'm not sure.

jkleiber, 2025-02-28:
Okay, let's keep this open then

linglejack06, 2025-02-28:
@aidnem  that was on a different branch without the fix, the lineup tune branch should have the fix but I want to check by on field and we haven't gotten to do that past two days

jkleiber, 2025-03-04:
@aidnem @linglejack06 I think the main issue is fixed but we should keep this open to figure out why OTF is bad on the first command in teleop

closed 2025-03-06 by aidnem