team401/2025-Robot-Code #89: Adding sensors and hardware to climber

minhnguyenbhs, 2025-02-13:
- [ ] find out what sensor + id is being added for detecting climb lock to cage and add it! 
- [ ] check talonfx variables + arm simulation values to match with reality

jkleiber, 2025-02-13:
If no sensor is used to detect that the cage is locked, this ticket will be used to figure out a way to make our own solution using current sensing

minhnguyenbhs, 2025-02-23:
Coach Preston said that there will be no sensor to detect cage locking- maybe should the code switch to seeing how the climb angle current changes to determine locking?

jkleiber, 2025-02-23:
Yes, but I imagine that this will start as a manual process. Let's get it working first and then we can collect some data on if using velocity or current would be better

jkleiber, 2025-04-01:
Closing this because there is no roadmap for adding a climb sensor for detecting cage lock. Our current system seems decent enough

closed 2025-04-01 by jkleiber