team401/2025-Robot-Code #90: Trigger warmup before lineup instead of when lineup begins

aidnem, 2025-02-13:
**Purpose**
Based on the current performance in sim, it seems that drive can line up much faster than scoring can warmup to score, especially in high positions such as L4. We should set a configurable distance away from lineup to begin warming up scoring.

**Project Scope**
- [ ] Make OTF state fire the BeginWarmup trigger when within a configurable distance of its goal

jkleiber, 2025-02-13:
@aidnem fwiw the sim performance is sus to me. In shop on Monday we were able to get to the top of the elevator's range in like 2 seconds. That being said, I don't think I have a problem with us getting the elevator up earlier provided we aren't in the middle of a dynamic maneuver while carrying algae or something

aidnem, 2025-02-13:
Okay, when I retuned I think I clamped the speed down too far I might just try letting it go faster and seeing what happens

aidnem, 2025-02-27:
@jkleiber Seems like this will be unnecessary based on how fast the elevator is now. We can reopen this if it seems needed.

aidnem, 2025-03-04:
@jkleiber hear me out: rather than warming up (we don't want the wrist to come out too early) we send the elevator to some setpoint, probably around halfway up, where we can balance having the robot be stable against having the elevator being close to its final position. It looked to me from our auto today that drive had to wait for scoring for a noticeable amount of time after it had finished lining up.

I might try writing code for this and then we only use it if we've exhausted our other options to improve autos

jkleiber, 2025-03-04:
Also can you send the video or something (email works or google drive)

jkleiber, 2025-03-04:
We should also probably get this branch merged and reset to a new integration branch

aidnem, 2025-03-04:
> We should also probably get the current branch merged and reset to a new integration branch

Agreed. Are you satisfied that lineup fix has actually fixed lineup? After we deleted all of the camera switching stuff it looked pretty reasonable to me. I'll send video along in an email in a minute.

jkleiber, 2025-03-04:
I'm convinced our OTF to lineup issues have been solved.

I'm not convinced that our OTF randomly failing in teleop issues have been solved

I also think we need to open a ticket to allow side step at the reef because I think we'll need that at comp. But if we don't have it at Portsmouth I won't lose sleep over it

closed 2025-03-04 by jkleiber