team401/2025-Robot-Code #98: Integrate wrist

aidnem, 2025-02-17:
**Purpose**
Integrate the wrist and tune it

jkleiber, 2025-02-18:
@aidnem after everyone left Preston tensioned the chain better. I think the gains we tuned will be fine for now, but we probably should consider adding a kG boost when we have algae detected in the claw (we have not tested this mode yet). As part of the PR associated with this ticket, we should figure out why pressing the Y button doesn't eject coral in test mode.

It would also be cool to have the gamepad joystick raise and lower the elevator setpoint rather than typing it into elastic (during setpoint tuning test mode). Then we could just have a readout on elastic of what the current values are so we can save them into our JSON files. If we can't get to this that's fine though - we just need to make sure we save working elastic dashboards to the robot / the repo

aidnem, 2025-02-18:
@jkleiber i think the limit switch stuff I was messing around with was probably why it wouldn't eject. I'll see if I can fix it next shop.

I'll see if I can make joysticks move the elevator setpoint a reasonable/controllable amount, that would be cool. Good idea

closed 2025-02-22 by aidnem