team401/2025-Robot-Code #100: Add ramp intake test mode

jkleiber, 2025-02-18:
**Purpose**
The purpose of this ticket is to add a test mode for the ramp intake. This will be used for tuning the ramp prior to full integration on the robot.

Ramp Intake tuning test mode
- PID gains (LoggedTunable)
- feedforwards (kS and kG for sure, kV and kA if we are using them) (LoggedTunable)
- setpoint entry (LoggedTunable)

**Project Scope**
- [ ] Add a ramp intake test mode to the ramp subsystem / test mode manager
- [ ] Test the test mode in 3D sim + on line graphs to show that it works
- [ ] Test on robot and determine the gains for the ramp

jkleiber, 2025-02-21:
@aidnem if you have time, can you take this issue on? I think we need to switch gears into "get things done now" mode and haven't heard back from @sleepyghost-zzz

aidnem, 2025-02-21:
Sure, my tickets are all either waiting on hardware testing or code review anyway.

aidnem, 2025-02-21:
@avidraccoon I added PID to this branch's test mode, if you don't mind I'll push and then it'll be there.

avidraccoon, 2025-02-21:
@jkleiber Do we need anything else currently or is just PID and setpoint good?

closed 2025-02-21 by aidnem