team401/2025-Robot-Code #101: Dynamically add keepout zones during auto

jkleiber, 2025-02-18:
**Purpose**
I am concerned that we will inadvertently run into the coral/algae stacks that are pre-staged near the alliance station wall during auto. To avoid running into this stuff, we should dynamically add in keepout zones around those things while in autonomous

**Project Scope**
- [ ] Add a keepout zone to the alliance side coral/algae pre-staged stacks when in auto
- [ ] Show it working in simulation

linglejack06, 2025-02-18:
@jkleiber i think the easiest way to do this is just to add a few blocks around those in the navgrid. otherwise, this would involve creating a custom pathplanner that we'd have to figure out how to interface with advantage scope with. 

the only problem is this would stay in teleop too. I don't see this being much of an issue though, as its such a small space that it won't really affect the paths efficiency

avidraccoon, 2025-02-18:
https://pathplanner.dev/api/java/com/pathplanner/lib/pathfinding/Pathfinder.html#setDynamicObstacles(java.util.List,edu.wpi.first.math.geometry.Translation2d)

linglejack06, 2025-02-18:
@avidraccoon good catch, but i think this method is not intended to be accessed in such a way, it looks like everytime we run a loop we would have to find the obstacles, pass them in, and pass the current pose in.

maybe im wrong though, im gonna look at advantage kits pathplanner

linglejack06, 2025-02-18:
maybe we could just set the obstacles at beginning of auto and teleop, but i dont think it works that way, as i think the obstacles passed in are only ones that would affect the current path

avidraccoon, 2025-02-18:
I looked at the code for LocalADStar you only need to set it once or if we want to move the obstacles.

avidraccoon, 2025-02-18:
As long as we are not making a new Object each time

linglejack06, 2025-02-18:
ok, thats helpful

linglejack06, 2025-02-18:
@avidraccoon im still skeptical though, if we only need to set it once, why
does the method even need the current robot pose

avidraccoon, 2025-02-18:
Because it internally converts the Pose to its corresponding Grid Position

PR #105 (2025-03-04): add dynamic coral algae stack obstacles
- add auto init and teleop init methods to drive to set dynamic obstacles

resolves #101