team401/2025-Robot-Code #102: Enhance the ramp subsystem

jkleiber, 2025-02-18:
**Purpose**
The ramp subsystem is very basic right now. I anticipate the following things will make this simplicity challenging to scale from:
1. There is a chance that we will need to maintain a third setpoint in case it is decided that the ramp should extend closer to the claw (and thus interfere with the elevator)
2. Based on some very low fidelity testing today, the ramp will likely need to hold a constant voltage while intaking (to mitigate coral recoil). We tested with no ability to do brake mode though

To handle these cases, we should do the following (see project scope)

**Project Scope**
- [ ] Add a state machine to the ramp subsystem with 3 states (Idle, intake, climb)
- [ ] In idle state, position control to near the intake position
- [ ] In climb state, position control to the climb position
- [ ] In intake mode, either do position control to the climb position or apply a constant voltage to drive the ramp against the hard stop
- [ ] Allow transitions between all states to all other states if triggered by externally commanded actions

jkleiber, 2025-02-21:
@linglejack06 I know this isn't your subsystem, but I think we need to get this done ASAP so I am assigning it to you

avidraccoon, 2025-02-21:
@jkleiber Would Something like this work?
```java

public class RampSubsystem extends SubsystemBase {

   private RampMechanism mechanism;
   private StateMachine stateMachine;

   public RampSubsystem(RampMechanism rampMechanism) {
      mechanism = rampMechanism;
   }

   public void fireTrigger(RampTrigger trigger){
      stateMachine.fire(trigger);
   }
   
   @Override
   public void periodic() {
      stateMachine.periodic();
      mechanism.periodic();
   }

   public boolean isInPosition() {
      return ((RampState) stateMachine.getState().getState()).inPosition();
   }

   public double getPosition() {
      return mechanism.inputs.position;
   }

   private StateMachine setupStateMachine(RampMechanism mechanism){
      StateMachineConfiguration config = new StateMachineConfiguration();
   
      config
         .configure(RampStates.IDLE)
         .permit(RampStateMachineTriggers.START_INTAKE, RampStates.INTAKE)
         .permit(RampStateMachineTriggers.START_CLIMB, RampStates.CLIMB);
   
      config
         .configure(RampStates.INTAKE)
         .permit(RampStateMachineTriggers.GOTO_IDLE, RampStates.IDLE)
         .permit(RampStateMachineTriggers.START_CLIMB, RampStates.CLIMB)
         .permit(RampStateMachineTriggers.HOLD_INTAKE, RampStates.INTAKE_HOLD);
   
      config
         .configure(RampStates.INTAKE_HOLD)
         .permit(RampStateMachineTriggers.GOTO_IDLE, RampStates.IDLE)
         .permit(RampStateMachineTriggers.START_CLIMB, RampStates.CLIMB)
   
      config
         .configure(RampStates.CLIMB)
         .permit(RampStateMachineTriggers.GOTO_IDLE, RampStates.IDLE)
         .permit(RampStateMachineTriggers.START_INTAKE, RampStates.INTAKE);
   
      StateMachine stateMachine = new StateMachine(config, RampStates.IDLE);
   
      RampState.setMechanism(mechanism);
      RampState.setFireTrigger(stateMachine::fire);
      
      return stateMachine;
   }
   
}

public interface RampIO {

   @AutoLog
   public static class RampInputs {
      public double position = 0.0;
      public boolean positionControl = false;
      public double controlValue = 0.0;
   }

   @AutoLog
   public static class RampOutputs {
      public double appliedVolts;
   }

   public void updateInputs(RampInputs inputs);

   public void updateOutputs(RampInputs inputs, RampOutputs outputs);
}

public class RampMechanism {
   RampIO io;
   RampInputsAutoLogged inputs = new RampInputsAutoLogged();
   RampOutputsAutoLogged outputs = new RampOutputsAutoLogged();

   private boolean positionControl = false;
   private double controlValue = 1.0;
   
   public RampMechanism(RampIO io) {
      this.io = io;
   }

   public void periodic() {

      inputs.positionControl = positionControl;
      inputs.controlValue = controlValue;
      
      io.updateInputs(inputs);
      io.updateOutputs(inputs, outputs);

      Logger.processInputs("ramp/inPosition", inPosition());
      Logger.processInputs("ramp/inputs", inputs);
      Logger.processInputs("ramp/outputs", outputs);
   }

   protected void setPosition(double position) {
      this.position = position;
      positionControl = true;
   }

   protected void setVoltage(double voltage) {
      this.voltage = voltage;
      positionControl = false;
   }

}

public enum RampStateMachineTriggers {
   GOTO_IDLE,
   START_INTAKE,
   HOLD_INTAKE,
   START_CLIMB
}

public abstract class RampState {
   private static RampMechanism mechanism;
   private static Consumer<RampStateMachineTriggers> fireTrigger;

   //If false controls the voltage
   private static boolean positionControl = false;
   private static double controlValue = 0.0;

   public static void setMechanism(RampMechanism mechanism){
      RampState.mechanism = mechanism;
   }
   
   public static void setFireTrigger(Consumer<RampStateMachineTriggers> fireTrigger){
      RampState.fireTrigger = fireTrigger;
   }

   protected void updateMechanism(){
      if (positionControl){
         mechanism.setPosition(controlValue);  
      }else{
         mechanism.setVoltage(controlValue);
      }
   }

   protected void setVoltage(double voltage){
      controlValue = voltage;
      positionControl = false;
   }

   protected void setPosition(double position){
      controlValue = position;
      positionControl = true;
   }

   @Override
   public void periodic(){
      updateMechanism();
   }
   
   @Override
   public void onEntry(Transition transition){
      updateSetPoint();
   }

   public boolean inPosition() {
      if (positionControl){
         return Math.abs(controlValue - mechanism.inputs.position) <= JsonConstants.rampConstants.positionRange;
      }else{
         return true;
      }
   }

}

public enum RampStates implements StateContainer {
   IDLE(new IdleState()),
   CLIMB(new ClimbState()),
   INTAKE(new IntakeState()),
   INTAKE_HOLD(new IntakeHoldState());

   private RampState state;

   @Override
   public StateInterface getState(){
      return state;
   }

   public RampState getRampState(){
      return state;
   }

   public RampStates(RampState state){
      this.state = state;
   }
}

public IdleState extends RampState {

   @Override
   public void periodic(){
      setPosition(JsonConstants.rampConstants.idlePosition);
      super.periodic();
   }
   
}

public ClimbState extends RampState {

   @Override
   public void periodic(){
      setPosition(JsonConstants.rampConstants.climbPosition);
      super.periodic();
   }
   
}

public IntakeState extends RampState {

   @Override
   public void periodic(){
      setPosition(JsonConstants.rampConstants.intakePosition);
      if (inPosition()){
         fireTrigger.accept(RampTriggers.HOLD_INTAKE);
      }
      super.periodic();
   }
   
}

public IntakeHoldState extends RampState {

   @Override
   public void periodic(){
      setVoltage(JsonConstants.rampConstants.intakeVoltage);
      super.periodic();
   }
   
}
```

jkleiber, 2025-02-21:
@avidraccoon at a high level this looks good. In the future, I recommend just creating a PR for this amount of code

linglejack06, 2025-02-21:
@avidraccoon @jkleiber i can write this code really quickly. but don't want
to take over from nathan since he already wrote the code above, what would
you prefer justin?

jkleiber, 2025-02-21:
@linglejack06 @avidraccoon I mainly just want a PR we can test in shop today. Can y'all discuss with each other and come to a decision?

aidnem, 2025-02-21:
By the way @linglejack06 @avidraccoon @jkleiber unless something has changed since #34, there won't be an encoder on the ramp so it will need to home to zero by itself. Whoever ends up doing this should take that into consideration.

closed 2025-02-28 by linglejack06