team401/2025-Robot-Code #119: Add button to confirm scoring when in manual mode

jkleiber, 2025-02-22:
**Purpose**
If we switch to manual mode in match, we should have a button to put us in warmup and another button for score

This way it's easier to manually lineup. Score wouldn't work until the scoring subsystem is in position 

As usual, this should be on the driver joysticks

**Project Scope**
- [ ] add button mapping for manual score

jkleiber, 2025-02-24:
@avidraccoon I've assigned this to you along with another manual autonomy ticket. What I would like you to do is the following:

1. Prioritize climb and ramp integration testing when possible
2. When not possible, I want to see some sort of design for the button bindings in each mode of operation. You can work that as either one of the tickets I've assigned you. Feel free to update one / combine them.
3. After you have a design for manual / mixed / full autonomy button bindings, please make a ticket or update this one to implement those bindings. Note: **only the driver should have bindings since the button masher is using snakescreen**. How you do this is the challenge to figure out. The driver should be able to do any binding without accidentally triggering another one, and while being able to drive at the same time.

aidnem, 2025-02-24:
@jkleiber @avidraccoon I was thinking about how to implement this and it's actually really simple: in manual mode, the "score" is essentially the driver telling the robot it is lined up so that the three conditions (elevator in position, wrist in position, drive in position) are all met. So: scoring already takes a supplier for whether or not drive is lined up. We should just make that supplier return: a) if drive is lined up, if in full or mixed autonomy or b) if the driver is telling us we're lined up, if we're in manual

jkleiber, 2025-03-01:
@aidnem please take this over, it HAS to get done asap

aidnem, 2025-03-01:
@jkleiber potential bindings (this requires rearrange some stuff but I think it's fine):

- Right joystick trigger: Warmup to score (This does OTF and then warms up when entering lineup in mixed autonomy, this just puts scoring into warmup mode when in manual autonomy)
- Right joystick top button: manually score in auto
- Left joystick top button: go to intake
- Left joystick trigger: hold to face reef automatically (I think this is a binding already but I'm not sure)
- Right joystick, button that's on top but below the "Top button" : seed odometry (not sure if this is even required this year, but if we need it it can go there)

jkleiber, 2025-03-01:
What is the top button? Is that easily reachable?

My understanding is that the bottom button on the joystick (what we've used for odometry reset last year) is way easier to press

Also "manually score in auto" - what does that mean

jkleiber, 2025-03-01:
Also feel free to rearrange bindings however you think is best, we literally haven't designed them at all

aidnem, 2025-03-01:
The "top button" I was referring to is the same button we used for odometry seed last year.

Sorry about the typo, my brain went faster than I could type for a second and I didn't notice, I meant to say "score in manual autonomy"

aidnem, 2025-03-01:
I'm going to make a google slide with the bindings as I make them so that we don't bind multiple functions to one button or make up buttons that don't exist (like I just did with the "button below top button")

aidnem, 2025-03-01:
[image]

Here's what I have so far, I guess either we remap buttons in endgame or we use those side buttons for climb.

jkleiber, 2025-03-01:
@aidnem this makes sense to me. I think for climb we only need one toggle button. It also can be somewhat harder to access since we don't want to accidentally deploy the climber

aidnem, 2025-03-01:
Sounds fair yeah. I'm working on digging through the bindings file right now. The actual scoring part of the bindings is really easy because it's just one change. I don't have joysticks at home so either I can try to mess with sim or we can just test tomorrow morning, it should take like 5 minutes

closed 2025-03-01 by aidnem