team401/2025-Robot-Code #120: Separate setpoints for close and far scoring

aidnem, 2025-02-22:
**Purpose**
Scoring needs to have different setpoints for scoring when not touching the reef (auto & mixed modes) vs touching the reef (manual)

**Project Scope**
- [ ] Add code to switch which setpoints smartly
- [ ] find the setpoints for both

jkleiber, 2025-02-23:
@aidnem I recommend closing this since aiming for a precise point in space results in limit cycles. Can revisit in the future if we have a specific need to do this for L1

aidnem, 2025-02-23:
@jkleiber Before I close this, I had one potential solution (for instance in case a coral gets stuck in front of the reef in auto, which would block us from ever scoring): If robot detects that along track is too large and not coming down, it could fallback to the far setpoints. We'd ideally never use them but it could be useful for unlucky times.

If we do go this route it would be pretty low priority since I'd rather be actually able to score than waste our time doing this and then not be able to score close or far away.

jkleiber, 2025-02-23:
Let's keep this issue open then to establish some lookup table based on along track/cross track distance 

It may be overkill but it could be something that opens up our tolerance on drive, so we should explore it if we have time in the future

jkleiber, 2025-03-11:
@aidnem @linglejack06 I've renamed this ticket and reprioritized it to Blacksburg comp. I think if we can create a lookup table of required wrist position based on along track distance we can go even faster in auto and teleop

I don't think we'll need to factor in robot velocity, but we will see when we actually test this on the robot

The way I see this working is some JSON lookup table (2 lists, one of along-track distance, another of wrist setpoints). Then we use interpolate double in coppercore to pick the wrist setpoint based on vision along track distance. We should have a feature flag for this, and maintain the constant wrist setpoint as a backup since we know it works at comp.

jkleiber, 2025-03-12:
Yeah we can see how it goes. If we can keep it a fixed height that is preferable though because it's simpler. I'm not sure how much height variation will buy us, but it might be worth drawing it out since this is a geometry problem

jkleiber, 2025-03-16:
Our reef at home is likely not accurate enough to do this tuning, and we are running up against time constraints for VABLA. Moving this to DCMP milestone to reduce schedule pressure

jkleiber, 2025-04-01:
wIth the new claw setup, this ticket is no longer relevant - closing

closed 2025-04-01 by jkleiber