team401/2025-Robot-Code #123: Fully implement snakescreen API

jkleiber, 2025-02-22:
**Purpose**
The purpose of this ticket is to get the software fully talking to snakescreen. 

We need robot software to be acting on the following info

- Reef Level (scoring subsystem)
- Intake Station Selection (drive / strategy manager?)
- Game Piece (scoring subsystem)
- Autonomy level (strategy manager / all subsystems with different behavior based on level)

We need robot software to publish the following information:

- Algae detected
- Coral detected

**Project Scope**
- [ ] Set up correct reads
- [ ] Set up correct writes

linglejack06, 2025-02-22:
@jkleiber this was partially solved in snakescreen integration. I think we still need to add game piece selection for net and processor and drive intake selection / intake mode if that's not gonna be a separate button on joystick

jkleiber, 2025-02-23:
Keeping this open until we fully implement the API, but the initial round of features worked well

linglejack06, 2025-02-23:
@PChild if i were to publish a scoring level on network tables, would Snakescreen update its ui accordingly? 

ie if robot default is level1 i want to publish that value so snake screen shows that robot hasnt been set to a different level after most recent code deploy

PChild, 2025-02-23:
> [@PChild](https://github.com/PChild) if i were to publish a scoring level on network tables, would Snakescreen update its ui accordingly?
> 
> ie if robot default is level1 i want to publish that value so snake screen shows that robot hasnt been set to a different level after most recent code deploy

It should work correctly. You should be able to test this in sim and verify.

closed 2025-02-28 by jkleiber