team401/2025-Robot-Code #131: Make scoring wait for velocity & position to score

aidnem, 2025-02-23:
**Purpose**
Scoring currently only waits until the wrist and elevator are within a certain margin of their goal position before it scores. This worked fine with "critically damped" (non-overshooting) controllers. After Justin's recent discussion post, we'll probably make the wrist and elevator controllers more robust, introducing slight overshoot. Because of this, scoring needs to wait for the elevator and wrist to be under a certain speed as well as being in position before it scores.

**Project Scope**
- [ ] Add speed thresholds for wrist and scoring which will affect wristReady and elevatorReady
- [ ] Test it in sim & on the robot

aidnem, 2025-02-24:
@jkleiber to deal with jittery velocity measurements, do you feel that it's better to median filter the velocity measurement or to manually calculate velocity using a filtered position measurement? (or something else)

aidnem, 2025-02-24:
Or we could just have a threshold high enough that jitter doesn't matter, I'll have to test this out to see how bad it is.

jkleiber, 2025-02-24:
Check to see how bad it is, but yeah a median or iir filter would be fine if filtering is required

closed 2025-02-28 by aidnem