team401/2025-Robot-Code #137: Pit system test mode

jkleiber, 2025-02-27:
## summary

We need a way to test the robot in the pit easily before every match

We should use test mode so the pit programmer can run to exercise all functions:
- test drive
- test elevator motion
- test wrist
- test claw
- test climb

This could be as simple as making a test mode that puts the robot in manual autonomy and then has bindings + elastic to move each subsystem appropriately 

Also we should document the system test procedure in a markdown file so the person running the test doesn't have to memorize what to do

## work scope
- [ ] add a test mode for system health checks pre match (preferably fairly automated so it's easy to run)
- [ ] add documentation on how to system test the robot prior to each match

Fyi: @aidnem I'm going to have @sleepyghost-zzz do this, so feel free to put your ideas for this in the ticket so he can implement them

aidnem, 2025-02-28:
Initial thoughts:

**How we should test each subsystem:**
First: open elastic, make sure there are no alerts from monitors tripping, this should be our first layer of defense.

- Drive: Command it to drive straight forward, then right, then make it try to rotate. It should probably do this at a slower speed and a faster speed so that we can see that all wheels point in the right direction and spin in the right direction.
- Elevator: Simple up-down motion, probably a pretty short range of motion so we don't flip the robot over in the pits or bonk the ceiling (maybe 0.5 meter travel or something?). This one will be pretty dangerous so definitely will need to be latched against some sort of manual confirmation to let it do its thing so we don't hit someone.
- Wrist: Move wrist through its full range of motion. This should be done in the middle of the elevator test (elevator up, move wrist from its top position to bottom, stop in the middle on the way up, come back to top). If it jitters or fails to control to its setpoint or anything, we know we have an issue.
- Claw: Run the rollers to verify that they can move. Then, separately (when robot isn't enabled) put a coral in the coral place and see if it logs that (we can put this boolean in the elastic layout for pit system test) and then do the same for algae.
- Ramp: Should move through a full range of motion. Up to the top, then down to the bottom.
- Climb: should be able to move up and down. This should happen during the ramp test, once ramp is up.

How this sequence would probably look:

- Make sure everybody is out of the robot
- Select system test in test mode selector.
- **Enable** test mode
- Elevator will home itself and then sit at zero.
- Press  "advance" button ->
- Drive will drive forward, sideways, then rotate. Verify all wheels orient correctly and drive in the same direction
- Press "advance" button ->
- Elevator moves up to 0.5 meters. Wrist swings down to its minimum angle, waits to reach setpoint. Then it comes up to its middle position, waits to reach setpoint. Then it comes back up to top. Watch wrist to make sure that it is controller properly, not stalling into hardstops, etc.
- Elevator moves back down to zero. Watch elevator and ensure that it controls correctly, etc.
- Press "advance" button ->
- Ramp moves up to top position. Climber will come up to the top of its range of motion, then down to the bottom. Make sure it works well.
- Ramp moves back down, make sure that it is working, controls properly, etc.
- **Disable** robot
- Place coral in claw by hand, verify that it shows up on elastic/snakescreen.
- Place algae in claw by hand, verify that it shows up on elastic/snakescreen.
- Check advantagescope or photonvision.local (TBD) to make sure that cameras are plugged in and working.

aidnem, 2025-02-28:
@jkleiber thoughts on how this could be improved (or if you think this will work)?

linglejack06, 2025-02-28:
this could esily be added with advance button in a similar way to how autos work in strategy manager if you need an example @sleepyghost-zzz

jkleiber, 2025-02-28:
Eventually we could probably have a snakesceeen / elastic maintenance page for testing an exact sequence and that way we'd be able to jump around tests and do auto advance. I don't think that will be in scope for Portsmouth, but feel free to make a ticket and we can get the Competition 2 Milestone going

aidnem, 2025-03-04:
Okay new plan:

- Elevator and wrist: setpoint tuning mode should work. Work needed: add the second masher joystick to control wrist angle in that test mode
- Ramp: ramp test mode. Work needed: buttons for ramp up/down in that test mode
- Climb: climb test mode.  Buttons already work. Work needed: Make those buttons only work in test mode, they do stuff in teleop sometimes
- Drive: teleop or a drive test mode could be used, tbh teleop seems fine
- Vision: photonvision.local is our friend. see if it can see tags, if it's blurry etc.
- Overall: check elastic for monitors

Reusing old test modes is much more achievable and much less of a waste of time imo.

@sleepyghost-zzz do you think you can knock these out soon?

sleepyghost-zzz, 2025-03-06:
yes