team401/2025-Robot-Code #138: Make competition representative autos

jkleiber, 2025-02-28:
## summary
We need some more autos besides "example auto". In this ticket, we should design which autos we want to run and then make our first set of options

I'm thinking we should have at a minimum:
- 1 piece auto where we score at the reef and then just sit there (we may need to update the json input to tell the robot to not intake in this scenario?)
- 2 coral auto on either side
- 3 coral auto on either side

Stretch goal
- 4 coral auto on either side

We may run out of time on the 4 coral auto due to needing to optimize lineup more. 

While there are some issues with auto in sim, this should be tested in sim + those issues should be worked around for now or fixed here. @minhnguyenbhs is working the longer term fix, so feel free to add hacks (in sim) to get things working here

aidnem, 2025-02-28:
@linglejack06 Have you encountered StrategyManager not scheduling the next action after the first score action completes? If so, have you found a solution? I'll keep working on trying to fix it but let me know if you know anything :)

linglejack06, 2025-02-28:
its probably because the intake action is never completing (like if it
never sees coral in sim its just gonna wait there i think)

aidnem, 2025-02-28:
After it scores, it never moves to go intake

aidnem, 2025-02-28:
@jkleiber for autos, do you feel like we should prioritize scoring easy scores for auto and then letting us do the harder, further away locations in teleop, or the opposite? I think that maybe maximizing the number of L4 in auto is what we want, since that will give us the most points (I'd rather do 4 L4 in auto and 3 in teleop than 3 in auto and 4 in teleop, for an example).

linglejack06, 2025-02-28:
@aidnem its probably waiting on scoring then before it moves on, like the scoring done supplier in drive lineup or in the auto score command

linglejack06, 2025-02-28:
Or it might be that auto score command waits on drive lineup state to go back to joysticks, if so that needs to be changed to rely on drive is lineup finished or whatever ( which might not be on main yet, I think that code is in lineup tune branch)

aidnem, 2025-02-28:
Have you ever tested out an auto intake command before? iirc scoring goes to idle and drive goes to joysticks mode but then it doesn't seem to move. I'm going to start debugging that first, because maybe it isn't a strategymanager issue.

linglejack06, 2025-02-28:
Also @aidnem why wouldn't we want to intake after last score? If we run out of time it will just get cleared anyways

aidnem, 2025-02-28:
I think it's more of a "something is horribly wrong with vision we can't drive reliably to the intake location" moment, we need to be able to not drive when we don't want to

linglejack06, 2025-02-28:
Yes I tested intake with drive and it worked in auto from scoring to intake. I think it's just the scori mg command is never finishing. Either scoring subsystem never reports done or drive is never reporting lineup finished

linglejack06, 2025-02-28:
Make sure auto intake command is firing drive otf trigger

aidnem, 2025-02-28:
Nvm I don't think drive is ever reporting that lineup is finished which is weird because it's letting scoring score :(

aidnem, 2025-02-28:
@linglejack06 found the issue, LineupState set driveLinedUp to false on exit, causing the score command to be unable to finish. Now it can intake and then gets stuck. Progress!

linglejack06, 2025-02-28:
Yeah that makes sense. Just set drive lined up to false on entry to lineup state please, that way it's still false and the command can run

linglejack06, 2025-02-28:
@aidnem are we gonna test this tonight? If so before we start I need to get lineup-tune merged (test in field ) so that doesn't cause issues

aidnem, 2025-02-28:
@linglejack06 2 things:

1)

> Yeah that makes sense. Just set drive lined up to false on entry to lineup state please, that way it's still false and the command can run

Done!

2) If we can be able to score in mixed autonomy (finish lineup-tune) I think this is the logical next step, although we also need to be able to test algae cycling (I think for now algae cycling will be fully manual, at least IMO auto/mixed algae scoring is a lower priority since it's less points).

aidnem, 2025-02-28:
I could also just rebase my branch off of lineup tune if necessary.

jkleiber, 2025-02-28:
@aidnem @linglejack06 we should only score on L4 for auto imo because then we don't have to worry about dislodging algae in auto and can get a lot of points early on

aidnem, 2025-02-28:
Agreed, also 7 points is crazy

aidnem, 2025-02-28:
@linglejack06 now it can score and intake in sim, except after the first intake is completed the action size goes to zero and it stops doing anything

jkleiber, 2025-02-28:
I'd check to make sure it isn't popping everything for some reason (more support for using a state machine because you could have more visibility into what's going on)

aidnem, 2025-03-01:
We fixed this at shop, I don't really remember what was wrong but somehow the auto score was saying it was finished instantly, causing it to instantly pop all items like you said.

closed 2025-03-02 by aidnem