team401/2025-Robot-Code #142: Add a gradle task to copy JSON files onto rio

aidnem, 2025-03-01:
**Purpose**
Deploying, even with just JSON changes, is still kind of slow. It will probably be faster to copy over the files and then restart robot code, because then the whole java compile step will be totally omitted (and it also probably won't have to deploy a JAR, not sure if it does this if the code was unchanged).

**Project Scope**
- [ ] Add a gradle task that copies over just the "deploy" folder
- [ ] Make it restart robot code so that constants are refreshed

aidnem, 2025-03-01:
@PChild do you know how to accomplish this? I've never manually worked on filesystem before on the rio

avidraccoon, 2025-03-01:
We could maybe look into the tasks that deploy to robot uses.