team401/2025-Robot-Code #144: Competition Compute Prep

jkleiber, 2025-03-02:
## Summary

- [ ] Make sure the backup beelink is ready
- [ ] Make sure the backup roboRIO 2 is ready
- [ ] Add a script for toggling overlayroot to both beelinks
- [ ] Turn on overlayroot on primary beelink

jkleiber, 2025-03-02:
@avidraccoon assigning this to you since you have some ideas about the overlayroot script. This needs to be done before comp, so preferably before wednesday

avidraccoon, 2025-03-04:
Is it fine if I use python to toggle the overlayroot

jkleiber, 2025-03-04:
@avidraccoon it's fine if it works and is easy to run

jkleiber, 2025-03-07:
@avidraccoon were you able to get the primary beelink set up with your scripts / make a coppercore PR with those scripts?

avidraccoon, 2025-03-07:
I have setup the beelink have not tested them to verify (Did not have time but that should take about 2-3 minutes at most)

And I was going to make a coppercore pr today

jkleiber, 2025-03-07:
Did you fix the issue with the backup beelink not being reachable at photonvision.local?

avidraccoon, 2025-03-07:
> Did you fix the issue with the backup beelink not being reachable at photonvision.local?

No, but I believe will has an Idea of how to fix it

jkleiber, 2025-03-24:
We had to do this in the pit at Blacksburg because the backup beelink was not fully ready smh. But now this is closed

closed 2025-03-24 by jkleiber