team401/2025-Robot-Code #145: Wrist climber avoidance

aidnem, 2025-03-03:
**Purpose**
The purpose of this addition is to make the wrist avoid the climber

**Project Scope**
- [ ] Add X

aidnem, 2025-03-03:
@jkleiber would you mind elaborating on how the climber will collide? I haven't seen the robot since it was put in its new position

PChild, 2025-03-03:
The bottom jaw of the claw can collide with the climber now that the climber is shifted back towards the elevator. We *might* be able to just stow the climber low enough that it's not a concern during general teleop play

aidnem, 2025-03-04:
Solved by stowing the climber at 0.7 radians during teleop, it's actually above the point where it would hit claw but below the point where it would climb by accident

closed 2025-03-11 by jkleiber