team401/2025-Robot-Code #146: Make reef side step work

jkleiber, 2025-03-04:
When we are lined up at the reef and want to switch to the spot adjacent to the current place we shouldn't need to back up. This ticket is to fix that

This won't go in before Portsmouth, the driver will just need to go to manual

Fyi: @aidnem

aidnem, 2025-03-14:
@linglejack06 @jkleiber could this be achieved by adding in alongTrack offset when crosstrack is above the "lined up" threshold? That way the robot wouldn't line up too soon when not sidestepping and it would also drive backward to sidestep using the same controller as normal lineup, right?

jkleiber, 2025-03-14:
@aidnem this might be as simple as making the camera switch work a little better and providing the driver a way to toggle between reef poles with a button

linglejack06, 2025-03-14:
I bet it's just messed up pid. Also a button would be much preferable to the logic that has to go on now with sidestepping. I feel like with how the logic is set up it will switch to other camera for one loop with the new reef location and the. Go back somehow.

Even though the reef location is then permanently set to the new one. I can mess with this in sim Aiden so you can work on the smart autonomy mode

jkleiber, 2025-03-14:
I think the PID gains are just too high for camera swap, and we'd need to use a different set of gains when on the "wrong" camera

Also any time we switch cameras we should reset the PID to ensure the integrator doesn't get wound up by big error

linglejack06, 2025-03-14:
If that doesn't make sense, the way logic works is it checks that reef side changes, then we will try the other camera for measuremebt

Then in the immediate next loop, lineup queries drive, which shows the new reef location now (sending correct camera index ) reef side check is false, so then we use the correct camera, and don't go through the try other camera thing.

At least this is how it should work, with what I saw last time we tested this in shop, it didn't seem like this was actually happening

jkleiber, 2025-03-14:
The sequence I'm imagining:

1. Driver lines up using right camera (smart auto maybe made this pick)
2. Driver wants to sidestep (maybe coral already scored in this spot)
3. Driver presses side step button 
4. Code wants to use left camera but only sees tag on right
5. Camera switch logic active, so we use lower gains
6. Robot drives right to get left camera centered on tag
7. Once tag in view on left camera, switch to our normal gains (correct camera mode)
8. Lineup proceeds as normal. Driver could press sidestep button again to switch back to the other pole theoretically

linglejack06, 2025-03-14:
> The sequence I'm imagining:
>
> 1. Driver lines up using right camera (smart auto maybe made this pick)
> 2. Driver wants to sidestep (maybe coral already scored in this spot)
> 3. Driver presses side step [...]

This makes sense I'll get it done

closed 2025-03-15 by jkleiber