team401/2025-Robot-Code #152: Make OTF not randomly fail

linglejack06, 2025-03-05:
**Purpose**
OTF sometimes tells us it's finished before even moving, fix this

aidnem, 2025-03-05:
Do we maybe want to try a minimum time? If the OTF command finishes in under (for example) 0.5 seconds from the time we enter OTF state, reschedule it?

Or if you have insight into why it's finishing that would be better. I'll test more autos in sim and see if I can figure it out too.

linglejack06, 2025-03-05:
@jkleiber cpuld it be this issue

https://github.com/mjansen4857/pathplanner/issues/1025

linglejack06, 2025-03-05:
We could be falling behind and not following quick enough, then it ends

jkleiber, 2025-03-05:
I don't think we're falling behind the pose. The robot literally is not moving in OTF despite the command being run in on entry

Let's just continually reschedule the command if we aren't within half a meter of the goal pose and log everything 

We should sim this and come to shop being able to see the goal pose, trajectory, and any exit criteria in scope

aidnem, 2025-03-05:
If it's instantly ending and it is that issue that would mean the target pose is never getting set, right? Otherwise I would think expected time would be non zero

jkleiber, 2025-03-05:
I think it should be fairly traceable with sim debugging. We need to log things and find the root cause

linglejack06, 2025-03-05:
> If it's instantly ending and it is that issue that would mean the target pose is never getting set, right? Otherwise I would think expected time would be non zero  

It is though, it's really weird. But I'm gonna try two ways to fix it: go back to vision, or just check if we are near the final pose (if we aren't reschedule)

Also adding this here for my own reference: forgot to push shop code, so first thing Wednesday we need to get hotel and push, still will be able to work in sim though

jkleiber, 2025-03-05:
I'll work on pushing the code since I'm still here

jkleiber, 2025-03-05:
@linglejack06 @aidnem code has been pushed