team401/2025-Robot-Code #156: Make ramp operational

jkleiber, 2025-03-06:
## Summary
The ramp is currently in-op. We need to make it have basic functionality before competition. No PID tuning is required or recommended since we have so little time remaining.

1. Make the ramp homing sequence faster
2. If the ramp goes too high above the homing area in idle, re-home to get back to the bottom
3. During climb, get the ramp out of the way. Once the ramp is out of the way, apply no voltage to the ramp

jkleiber, 2025-03-06:
@avidraccoon yes. Also similar in idle: if the ramp goes too far away from intake then we'd want a voltage to be applied to make it go back down

aidnem, 2025-03-06:
@jkleiber @avidraccoon  If we put the ramp motor in brake mode it would also probably protect us against obliterating the top of the ramp on the hardstop when it goes to climb position. Then we could literally have the climb state just be: if position is below a certain spot (wherever the center of mass goes onto the other side of the pivot) apply some big voltage, then brake mode will catch it and not let it slam on the way down the other side.

jkleiber, 2025-03-06:
Sure that works

We also need to make sure we can't raise the elevator into the ramp / smash the ramp into the elevator on accident if extended. But I'm fine with making the driver manage that for now and we add that for Blacksburg

aidnem, 2025-03-06:
That'll be a 1-line change in scoring subsystem, I'll make a branch for this and knock it out rn so that @avidraccoon can make any necessary ramp changes on top of it.

closed 2025-03-08 by minhnguyenbhs