team401/2025-Robot-Code #167: Fix auto lagspike

aidnem, 2025-03-09:
**Purpose**
Stop auto from taking 1.5-2.5 seconds to run the first cycle on auto init which makes the difference between 2 and 3 piece auto

**Project Scope**
- [ ] Find if the issue happens in sim
- [ ] Fix the autoInit function taking forever (>0.5 seconds)
- [ ] Fix the first cycle taking forever (>0.8 seconds)
- [ ] Fix the pathplanner command first execute taking forever (>0.8 seconds) (Probably part of the above)

aidnem, 2025-03-09:
[image]

It does happen in sim !

aidnem, 2025-03-09:
Moving dynamic obstacles to drive init doesn't really do anything :(

[image]

jkleiber, 2025-03-09:
If we OTF just randomly in teleop does this happen?

aidnem, 2025-03-09:
Not sure, I'll test that next. I did discover that every time we enable auto, it does this (but less bad):

[image]

(that's 769ms periodic, then 640ms, then 605)

I'll test tele next.

The screenshots above are with me manually creating an OTF command and scheduling it in Drive init (and then cancelling it in autonomousInit function).

aidnem, 2025-03-09:
We get intermittent spikes to 60ms during OTF in teleop, so I don't think OTF is actually the problem.

The spike happens every time we enable auto OR teleop :(

[image]

(here's me switching between the two)

aidnem, 2025-03-09:
I think we're completely dodging the pathplanner command lag issue in sim because the actual match log shows the teleop init lag but it's much less significant than the auto init lag:

[image]

I'm going to try profiling strategy manager and see if it's taking a forever to schedule stuff or something silly like that. I think moving the command might fix this but I'm really not sure. I'll trace more init methods and see what we're even doing.

PChild, 2025-03-09:
Can we also check the brake mode and LED setting calls while we're profiling stuff? Those are common to both auto and tele init. They shouldn't be slow but why not check, right?

aidnem, 2025-03-09:
Ok are we sure there isn't like some sort of inherent lag in enabling?
All 4 things took 0.0 seconds:

[image]

And yet:

[image]

TeleopInit somehow took 0.62 seconds???

aidnem, 2025-03-09:
I think sim is wrong and the lag we're seeing in sim is only partly related to the lag on the real robot. I put a fake OTF command in drive init and hopefully that'll shave 0.9s from our first cycle time in auto.

jkleiber, 2025-03-09:
@aidnem I think we should comment out the set subsystem to brake function call in auto/teleopInit. Maybe applying configs takes a long time and is blocking (wouldn't surprise me)

aidnem, 2025-03-10:
Final thing I'd like to check is if disabling hoot logging from phoenix 6 will eliminate the final 300ms delay. 0.3 seconds is still a lot, and it will definitely make the difference between scoring a piece and not scoring it in auto.

[This CD post describes the issue](https://www.chiefdelphi.com/t/auto-init-overrun/494848/14)

aidnem, 2025-03-12:
@linglejack06 we got the lag down to 300ms, I want to try disabling hoot logging and see if that removes the last 0.3 seconds.

aidnem, 2025-03-13:
Whenever we decide to try to remove the last 300ms: [https://www.chiefdelphi.com/t/auto-init-overrun/494848/32](https://www.chiefdelphi.com/t/auto-init-overrun/494848/32).

I think 1 will be an easy thing to try & test at shop. 3 is another one that should be fairly trivial to implement. 5 would be a good last resort to try but it seems like it might do suspicious stuff to other threads. We've already done 4 and 2 should be tested at competition since we don't have FMS at home.

jkleiber, 2025-03-16:
we should do this alongside #191 , so moving this to Comp 2 Readiness milestone