team401/2025-Robot-Code #168: Less aggressive OTF and warmup in teleop

aidnem, 2025-03-10:
**Purpose**
OTF drives really hard and warms up too early in teleop.

**Project Scope**
- [ ] Separate OTF gains for auto and teleop
- [ ] Only activate far warmup and early warmup when in autonomous; these features activate way too early when defended in teleop

jkleiber, 2025-03-10:
Agree with this. As part of this issue we need to also avoid slapping algae off the reef because that was called as a foul at vapor due to using coral to dislodge algae (incorrect call imo, but need to not let the refs make that decision)

aidnem, 2025-03-11:
How did we even slap algae off the reef? I couldn't see very well because I was behind the refs

jkleiber, 2025-03-11:
I think the wrist would tilt down and then we'd hit the algae from the bottom. It happened in our first playoff match near the end of teleop, might be worth looking it up on TBA to see if that is helpful when comparing to logs

aidnem, 2025-03-11:
I'll look it up, thanks. I won't be at shop even if it is open until Sunday/Monday so I'd need to get someone else to download logs and push comp changes if we want to do that before then

aidnem, 2025-03-14:
Looks like that was the match where the elevator was tilted forward further than normal. I can still clamp it harder but this shouldn't be as much of an issue if our robot is correct geometrically.

jkleiber, 2025-03-14:
Okay then probably no change required on that, thanks for looking into it

aidnem, 2025-03-14:
I found the logs to be sure:

[image]

This is the last time the elevator goes up in periodic (green line) and the wrist angle (gray line). It looks like the impact occurs at about 0.25 rotations or 1.59 radians. This is lower than the idle setpoint of 0.32 rotations, so the clamps *could* be more aggressive if we want to err on the side of safety. Your call @jkleiber, the video can be seen [here at 2:30s](https://www.youtube.com/watch?v=CKdO6QSgYa0&t=150s).

jkleiber, 2025-03-14:
If we make the clamps more aggressive, what are the knock on effects to the system?

If the elevator was out of tolerance then I don't want to change clamps based on that.

aidnem, 2025-03-15:
I agree. We already need to shave time off auto and i think more aggressive clamps would slow us down slightly.

jkleiber, 2025-03-16:
This is a nice to have, but is not strictly required for VABLA. Moving to DCMP milestone to make that prioritization clear