team401/2025-Robot-Code #169: Add individual reef pole OTF poses

aidnem, 2025-03-10:
**Purpose**
Lineup is slow when crosstrack error is large; having 12 OTF poses instead of 6 would greatly speed up our autos and our teleop cycles

**Project Scope**
- [ ] Add separate OTF poses for each reef pole
- [ ] Find each pose for constants
- [ ] Test in sim & on real robot

jkleiber, 2025-03-14:
@linglejack06 can we prioritize this one? I think we don't really need the poses to be picked in shop, as we only have 1 side of the field anyway

We can edit the ones picked from sim as needed in shop

aidnem, 2025-03-14:
@linglejack06 You can also use pathplanner to find the poses and it should be very accurate. Let me know if you want me to take this, I've already found 36 other poses with that method in the last 24 hours :sob:

linglejack06, 2025-03-14:
> @linglejack06 You can also use pathplanner to find the poses and it should be very accurate. Let me know if you want me to take this, I've already found 36 other poses with that method in the last 24 hours :sob:

I can get that done, parhplanner sounds a lot easier. Do we want to keep the poses the same distance away from the tag?

aidnem, 2025-03-15:
@linglejack06 I think keeping them the same distance away will be good. Lineup looked like it worked really well (except for crosstrack taking a while), so probably keeping it similar would be good. @jkleiber agreed?

jkleiber, 2025-03-15:
Agreed

closed 2025-03-15 by jkleiber