team401/2025-Robot-Code #174: Score on nearest pole in mixed autonomy

aidnem, 2025-03-12:
## Summary

We had driver complaints about how selecting a reef location on snakescreen was too slow and required too much communication. We should instead pick the nearest reef pole at the time the OTF trigger is pulled and drive to that and score.

## Project Scope

- [ ] Select nearest reef pole in mixed auto
- [ ] Test in sim

aidnem, 2025-03-12:
@jkleiber does this sound like what you and oliver were discussing for a better operator interface?

jkleiber, 2025-03-12:
Yeah pretty much @aidnem 

In certain scenarios we should skip OTF and go directly to lineup. For example, if we are already up against the reef

This should be a new autonomy level in order to avoid removing the existing functionality (which may be useful as a fallback plan)

aidnem, 2025-03-12:
Sure, I'll rename this ticket accordingly

aidnem, 2025-03-12:
Done. @jkleiber I also added that this should be able to automatically select an algae intake level based on the current position of the robot, since (as far as I could tell from reading the game manual last night) the algae locations are fixed.

jkleiber, 2025-03-12:
Yeah that makes sense, thanks

My mental model for what we should do for picking the best reef location is as follows:
1. If we can see a tag with a camera and are less than TBD distance away, just go to lineup and pick that camera and tag
2. If 1 does not apply, use global pose to pick which pole is closer (pole, not OTF location). OTF to that pole's OTF location and enter lineup like we normally do

Also a drive by change of this PR might need to be that if we get into the other camera during lineup we reduce the gains a bit on crosstrack because insane driving happened due to large error during comp in that case. Maybe that's a separate ticket for @linglejack06 though

jkleiber, 2025-03-12:
Made #175 for that last part

aidnem, 2025-03-13:
Yeah I'm thinking like 99% of this will be the button binding InstantCommand calling a StrategyManager (or Drive) method to figure out which reef location is closest to score on. Beyond that it should function the same as mixed auto.

closed 2025-03-15 by aidnem