team401/2025-Robot-Code #178: Bounce ramp at the start of match

aidnem, 2025-03-13:
**Purpose**
The ramp has a new extending mechanism that has to be folded in at the start of a match in order to fit within frame perimeter. At the start of a match, the ramp must bounce up to let this mechanism unfold and extend.

**Project Scope**
- [ ] After homing, bounce the ramp up a little bit (maybe add a state that just controls it up to a small angle above minimum and then goes to idle after a certain period of time. This state could be transitioned to from the homing state).

aidnem, 2025-03-13:
@PChild Please let me know if this isn't what you need, I'm going off of a vague description from oliver

PChild, 2025-03-14:
This is a perfect description of the requirements, we finished up the first rev of hardware for this tonight and can test it whenever programming is ready.