team401/2025-Robot-Code #191: Auto optimization (VABLA)

jkleiber, 2025-03-16:
## Summary

On 3/15 we did a 3 piece auto in shop and had the following total time allocated to the following tasks:
- OTF to reef: 3.639 sec
- OTF from reef to coral station: 3.475 sec
- Lineup at reef: 2.746 sec
- HP reaction time: 0.868 sec
- HP coral release to coral in claw time: 1.601 sec
- Waiting for wrist to be in position: 0.4 sec
- Robot sitting still between actions: 1.304 sec
- Scoring coral: 0.427 sec
- Assumed 0.3 second delay
Total time: 14.76 seconds

We can influence the following:

1. We should reduce that 0.3 second delay at the start of auto
2. Reef lineup is taking a long time because we're over shooting our OTF targets. If we lower OTF accel maybe we will actually go faster by spending less time lining up
3. We need to look into why we're sitting still between actions - is this an OTF thing? It's maybe 0.25 seconds per action, but that adds up a lot
4. It's worth figuring out why we take nearly the same amount of time to go to the intake as it does to go to the reef, even though we go to the reef 3 times vs 2 times to the coral station. Maybe the long drive at the start could have a faster max velocity
5. Wrist not being in position due to reef avoidance is sad. From watching the video it looked like we would've been fine so maybe we just need to lower the L4 reef avoidance height

aidnem, 2025-03-16:
I'll look at logs and see if I can figure out why it's sitting in between actions. Doesn't really make sense to me, maybe it is an OTF lagspike issue, or commands taking a few cycles to get scheduled?

aidnem, 2025-03-16:
Upon examining logs, Q54 (our best auto at comp) had lagspikes of a few 100 MS right after scoring each piece in auto:

It looks like robotPeriodic is taking a long time right after a score:

[image]

And also sometimes randomly (I can't figure out anything that should've caused this one: drive state remained the same, scoring state remained the same). Maybe GC?

[image]

However robotPeriodic took 76MS, probably causing the spike:

[image]

jkleiber, 2025-03-16:
@aidnem I could be wrong but this looks like an OTF thing. 

If we can score a coral while planning the next OTF path (despite the lagspike) we'd likely not notice this. Scoring a coral takes over 100ms so if we could parallelize that somehow we'd gain a bunch of time back. Similarly for intake, if we can plan the OTF without running it until we get the coral, we'd save a bunch of milliseconds

aidnem, 2025-03-16:
I agree, that's definitely a lot of it (I saw some console messages about pathfind command execute taking 5ever but not always).

Here's my thinking after spiraling for a while (wasn't sure if there was an actual way to warmup so I thought worst case, we just manually call execute and then override the output):

- The command needs to be warmed up after scoring applies outputs for the first time in score state. This is because the rollers need to be spinning already when the lagspike occurs.
- Command needs to either have its output disabled or be executed before the state machine periodic so that whatever crazy goalspeeds it commands are immediately replaced by the correct behavior from the lineup/joystick state

How to accomplish this:

- Scoring: keep track of whether it's the first loop of scorestate (keep a boolean). On the first loop of scorestate, call a callback that sets a flag to true in drive.
- When that flag is true in drive, the first thing it should do in periodic is call a static method on OTFState to warm up its next OTF command. Then, the state machine periodic will destroy any bad outputs generated by executing the command.

The drive side of this is a little gross so I'm going to read docs and see if there's a way to properly generate the path before actually applying goalspeeds with the command.

linglejack06, 2025-03-16:
> I agree, that's definitely a lot of it (I saw some console messages about pathfind command execute taking 5ever but not always).
>
> Here's my thinking after spiraling for a while (wasn't sure if there [...]

@aidnem why don't we just generate command and just not schedule it yet??

It would still run through processing then but wouldn't send any bad speeds, this would avoid a stupid hack in set goal speeds

linglejack06, 2025-03-16:
Just saw your comment at the bottom, I think the way to generate the path is just run the pathplanner.pathfindto tjing that we already run (then we just don't schedule it yet) 

so just make the get drive to pose method in otf static and call it while scoring

aidnem, 2025-03-16:
Wasn't sure if generating the command caused it to actually do the pathfinding lol. If it does then dont do the garbage hack. Does it pathfind when created?

linglejack06, 2025-03-16:
That would be the logical thing and what I can discern happening based on looking at the class

But not 100% sure, might as well try before doing a stupid hack, I can do this in class tmr

jkleiber, 2025-03-17:
For what it's worth, you also don't need to track the first loop of score state - you can just use onEntry

aidnem, 2025-03-17:
@jkleiber it we cause a large lag spike in onEntry, applyOutputs won't have run yet and we won't actually be applying power to score the coral while the lag occurs

jkleiber, 2025-03-17:
If we are single threaded (we run odometry on one thread, and the main robot software on another), we would be able to send drive a trigger in score onEntry because then drive would only run after that score state / scoring subsystem runs though?

In any case I think it's fine to do it in periodic and track the number of loops

aidnem, 2025-03-17:
I think you're actually right. The trigger to send scoring to score state will be called when scoring state machine periodic happens since it's fired by warmup state periodic. As long as it tells drive to warm up as soon as possible and doesn't instantaneously start the warmup, claw mechanism apply outputs will be run since it's right after state machine in scoring periodic.

Sorry, I had to fully think it through before I realized

jkleiber, 2025-03-17:
If y'all can get a branch put together that attempts this performance boost before shop it will be our top testing priority for tomorrow. The way I see it is that if we can even cut the time sitting idle in half, we have a full second remaining after the 3rd coral

Then we can start playing games with OTF and lineup to see how much time we can cut from there. Also, we should attempt to remove the 0.3 second spike as part of the above (compute the first auto OTF path in disabled periodic?)

Savings goals:
- remove 0.3 initial spike (-0.3)
- cut idle time in half (-0.7)
- slower OTF to reef (+0.5)
- faster OTF to coral station (-0.5)
- better lineup (-1.0?)

That would put us at 12.76 seconds for 3 coral, and we'd probably be hitting a 3.5 coral auto. If we get luckier with intake it's 11.76 with an outside chance at 4

aidnem, 2025-03-17:
@linglejack06 do you want to work on this first or should I? I have a study hall third block so if I don't hear from you before then I'll knock it out then. I can also work on the branch after you work on it if you want to do some stuff before then and then have me do more.

My thinking is, we could have strategy manager update drive with a "next desired location" (or directly update OTFState with some static field) and then scoring would just trigger the warmup whenever it was ready.

linglejack06, 2025-03-17:
ive already added a way for score to warmup next command (but right now it still utilizes the previously set pose)

jkleiber, 2025-03-18:
According to the time analysis from yesterday's autos, we basically saved no time overall LOL. We did save small amounts of time in certain areas (wrist, HP, lineup), but lost time in other areas (first coral OTF mainly)

We have the following untested things we should try to test today in at least 1 auto run:
1. Higher elevator kP
2. Disabling signal logger 

If we have additional time today (after testing climb), we maybe try the following:
1. @linglejack06 's idea to OTF into the exact lined up position has some merit because then we can't overshoot much (we will hit the reef). Specifically we should try this on the first reef pole we go to in auto (Reef3) because that one had worse performance than Saturday by a lot.
2. If that doesn't work, we should pick a reasonable option for OTF to reef 3 and then merge this branch. If it does work, we can get good OTF poses for the other poles and then merge this branch, with follow on work in sim/path planner to get the other poses.

We should focus purely on shaving time off of the first coral placement, which is our main problem right now.

aidnem, 2025-03-18:
@jkleiber third untested thing: I (hopefully) fixed OTF warmup code so if that actually would've saved time it will only apply today, I realized we were regenerating them in OTF onEntry in yesterday's code.

closed 2025-03-19 by aidnem