team401/2025-Robot-Code #197: Add ability to have intake and score algae locations

linglejack06, 2025-03-19:


linglejack06, 2025-03-20:
@jkleiber thsi doesn't relate to this ticket but don't forget to push the changes

jkleiber, 2025-03-20:
@linglejack06 @aidnem pushed changes, @PChild and I did some mentor built snake screen stuff in the climb branch but we had problems with NT seemingly ignoring algae mode again. Maybe give it a try in sim and see if it's just a networking issue?

aidnem, 2025-03-20:
@PChild would re initializing all of the subscribers (and maybe publishers) in teleop init fix this issue if removing default value setting from Snakescreen doesn't?

avidraccoon, 2025-03-20:
@aidnem At least with the way I understood what Preston said I believe that would just make it worse.

PChild, 2025-03-20:
That would not help, I don't think. What I would like is for you all to add a function that sets sane defaults on teleInit: L4 coral, Smart autonomy, Reef algae score, coral mode, maybe which intake station

PChild, 2025-03-21:
@avidraccoon can you push code from tonight please

linglejack06, 2025-03-25:
@jkleiber wasn't this issue partially solved late last week? if so, what is left to add, i can open a PR with what is left

jkleiber, 2025-03-25:
@linglejack06 I don't think we fixed this since the OTF positions were on the left(?) for every algae intake. We'd want them in the center

Idk if we are doing lineup after algae otf, but if we are then we need to add a camera offset for that to make it work

jkleiber, 2025-03-25:
@linglejack06 yeah we need to be centered because by the time we get to the OTF pose, we are nearly touching the algae with the claw.

I think both cameras should be able to see the tag sonce we'll be far enough away. Do we have an along track offset for algae as well?

linglejack06, 2025-03-25:
theres no along track offset, since i figured it would be the same as coral (against the reef)

my only concern with lineup is that once we get close enough it might lose measurements when centered

aidnem, 2025-03-25:
@linglejack06 We should switch this to linear drive as soon as that merges because then we can reliably enter lineup without insane velocity

jkleiber, 2025-03-25:
@linglejack06 I don't think the robot is pressed up against the reef during algae intake. We could check match video to confirm though

linglejack06, 2025-03-28:
> @linglejack06 <https://github.com/linglejack06> based on shop last night
> when we got stuck in algae mode by accident (scoring being null defaulted
> to that; it's fixed now), the crosstrack offsets on that branch were in the
> wrong direction. Instead of driving toward the center of the reef, it was
> driving away from the center and then losing the tag. I'm not sure if you
> already fixed this but it's something to keep in mind.

ok, thats easy to fix in json constants, i can push that change