team401/2025-Robot-Code #199: Go to closest position on coral station

aidnem, 2025-03-20:
**Purpose**
Sometimes, driving to one specific point on the coral station will be slower than picking the closest point in on the coral station. Potentially, when after scoring position 5, it would be faster to drive straight back instead of strafing to the same corner of the coral station that we are currently using.

Instead, we should keep track of a pose at each end of the coral station and then OTF to the closest point along this line whenever we want to go to intake. This will also enable smart mode to be used in teleop more  when playing against defense, since it will OTF to the closest point on the coral station.

**Project Scope**
- [ ] Add a translation for the opposite end of each coral station (the driver station wall side), make the OTF pose a translation (for the opposite side), and add a rotation2d for orientation.
- [ ] Add a way to get the closest point along the line between the two translations (maybe a new coppercore feature?)
- [ ] Generate a pose using the closest translation on the line and the rotation from constants when generating an OTF path to coral station.
- [ ] Keep track of the desired location that the OTF path was generated for to check if the OTF path is correct in OTFState periodic (Currently it verifies that the target pose matches the target pose of the current DesiredLocation, but since this pose will be variable, that will result in constantly regenerating the path).

avidraccoon, 2025-03-20:
I think this function should work for getting the closest point between the Translations
```java
public Translation2d closestPointOnLine(Translation2d linePoint1, Translation2d linePoint2, Translation2d point){
	double pointX = point.getX();
	double pointY = point.getY();
	double linePoint1X = linePoint1.getX();
	double linePoint1Y = linePoint1.getY();
	double linePoint2X = linePoint2.getX();
	double linePoint2Y = linePoint2.getY();
	double distX = linePoint1X - linePoint2X;
	double distY = linePoint1Y - linePoint2Y;
	double lineLength = Math.sqrt(distX*distX + distY*distY);
	double dot = (((pointX - linePoint1X) * (linePoint2X - linePoint1X)) + ((pointY - linePoint1Y) * (linePoint2Y - linePoint1Y))) / Math.pow(lineLength, 2);
	double closestX = linePoint1X + (dot * (linePoint2X - linePoint1X));
	double closestY = linePoint1Y + (dot * (linePoint2Y - linePoint1Y));
	double constrainedX = Math.min(Math.max(closestX, Math.min(linePoint1X, linePoint2X)), Math.max(linePoint1X, linePoint2X));
	double constrainedY = Math.min(Math.max(closestY, Math.min(linePoint1Y, linePoint2Y)), Math.max(linePoint1Y, linePoint2Y));
	return new Translation2d(constrainedX, constrainedY);
}

```

aidnem, 2025-03-21:
Sorry I'd need to learn the math behind it to understand it but based on the one stack overflow answer I've read I think so

avidraccoon, 2025-03-21:
@aidnem https://www.desmos.com/geometry/biweb4ikhf

aidnem, 2025-03-21:
Woah that's awesome tysm

aidnem, 2025-03-21:
@avidraccoon Yes, that method looks like it works. Thanks for the desmos link, that made it make sense.

avidraccoon, 2025-03-21:
Does the method look to implement the math correctly?

avidraccoon, 2025-03-21:
I think my math might be off, for the one using the vectors

avidraccoon, 2025-03-21:
I think I implemented the math wrong

aidnem, 2025-03-21:
Does desmos `vector(B, A)` mean the same thing as `linePointBVector.minus(linePointAVector)`?

avidraccoon, 2025-03-21:
I convert the vector(B, A) the wrong way

avidraccoon, 2025-03-21:
@aidnem This should be right I think
```java
public Translation2d closestPointOnLine(Translation2d linePoint1, Translation2d linePoint2, Translation2d point){
	Vector<N2> linePointAVector = linePoint1.toVector();
	Vector<N2> linePointBVector = linePoint2.toVector();
	Vector<N2> pointVector = point.toVector();
	Vector<N2> lineVector = linePointAVector.minus(linePointBVector);
	Vector<N2> pointLineVector = linePointAVector.minus(pointVector);
	double t = pointLineVector.dot(lineVector) / Math.pow(lineVector.norm(), 2);
	t = Math.min(Math.max(t, 0.0), 1.0);
	Vector<N2> resultVector = linePointAVector.plus(lineVector.times(t));
	return new Translation2d(resultVector);
}
```

avidraccoon, 2025-03-21:
wait forgot one part

avidraccoon, 2025-03-21:
@aidnem This should be right
```java
public Translation2d closestPointOnLine(Translation2d linePoint1, Translation2d linePoint2, Translation2d point){
	Vector<N2> linePointAVector = linePoint1.toVector();
	Vector<N2> linePointBVector = linePoint2.toVector();
	Vector<N2> pointVector = point.toVector();
	Vector<N2> lineVector = linePointAVector.minus(linePointBVector);
	Vector<N2> pointLineVector = linePointAVector.minus(pointVector);
	double t = pointLineVector.dot(lineVector) / Math.pow(lineVector.norm(), 2);
	t = Math.min(Math.max(t, 0.0), 1.0);
	Vector<N2> resultVector = linePointAVector.plus(linePointBVector.minus(linePointAVector).times(t));
	return new Translation2d(resultVector);
}
```

aidnem, 2025-03-24:
@jkleiber @avidraccoon Since we preferably need to line up with a slot when intaking, we should take the number of slots between the two setpoints and then round our OTF pose to the nearest slot position. This will make our human player's life easier and speed up our intake in auto IMO.

avidraccoon, 2025-03-24:
public Translation2d closestPointOnLineRounded(Translation2d linePoint1,
Translation2d linePoint2, Translation2d point, int divides){
Vector<N2> linePointAVector = linePoint1.toVector();
Vector<N2> linePointBVector = linePoint2.toVector();
Vector<N2> pointVector = point.toVector();
Vector<N2> lineVector = linePointBVector.minus(linePointAVector);
Vector<N2> pointLineVector = pointVector.minus(linePointAVector);
double t = pointLineVector.dot(lineVector) / Math.pow(lineVector.norm(), 2);
t = Math.round(divides * Math.min(Math.max(t, 0.0), 1.0));
Vector<N2> resultVector = linePointAVector.plus(lineVector.times(t));
return new Translation2d(resultVector);
}

> Message ID: ***@***.***>
>

jkleiber, 2025-03-24:
@avidraccoon if you're going to write code for this or #199, please make a PR and link it to the ticket. It's hard to review code in issue comments and code can't be useful without context anyway

jkleiber, 2025-03-25:
@aidnem I think the most important thing regarding shaving off time in auto is for us to go to the same coral station position every time (so the HP doesn't have to think about where to be more than one time). So I think this should probably only be active in teleop at first

I'm going to move this off the DCMP milestone since it's not **technically** required, but if we do get this done and tested it will go in before DCMP if that makes sense. Same disposition for #188