team401/2025-Robot-Code #203: Linear point to point driving

jkleiber, 2025-03-24:
# Summary 

In auto it might be faster if we made our own motion profiled point to point "linear drive"

We should be able to leverage the WPILib trapezoidal profile here. 

The concept is to command a velocity vector that points from robot position to a goal pose. The velocity magnitude would be based on distance and trapezoidal profile

This would be mainly used in auto as an alternative to OTF to increase driving stability

jkleiber, 2025-03-27:
@aidnem @minhnguyenbhs 

Thought about this on the ride home: what if we split this into 2 phases

1. Before we are within X degrees of the target position (with respect to the desired heading at the end) drive to a point Y meters behind the target point (projected backwards along target heading)
2. Once within X degrees, drive to the target point

I think this will make it so we can have an ending velocity while smoothly entering lineup. Thoughts?

aidnem, 2025-03-27:
@jkleiber this sounds like it might work. My only concern is that it will overshoot its target heading (I'm imagining like a pizza slice shaped area in which it can actually drive to the correct target pose) and have to correct back the other way, or that maybe we won't be able to accelerate as fast as we need to be able to in order to change direction instantly like that.

I also thought about this for a while and realized, I don't think we need a PID controller for translation (and maybe rotation as well but I'm not as sure about that one). We know our acceleration, max velocity, target position, etc. Drive already does PID control to take a set of goal speeds and control the drivetrain to those speeds. The way I see it, we should just use a trapezoidal profile: calculate it based on distance to target with zero as our goal, then use that target velocity to generate goal speeds that have direction pointed toward wherever we're going (or behind it if we do the 2 phase system) and a magnitude of the velocity we get from trapezoid profile. Do you foresee any problems with this? (you too 
@minhnguyenbhs)

jkleiber, 2025-03-27:
I think we don't need a PID either as long as we update a trapezoidal profile each loop

I'm not worried about acceleration in the 2 phase system because we'd just need to make sure we pick good values for the slice angle and length. To make sure we don't get cooked by the thin part of the slice, we could ignore the slice logic and just drive to the final target once we're within Y meters of the final target

jkleiber, 2025-03-27:
We will need PID for the rotation though imo

closed 2025-03-31 by jkleiber